SeedSigner/seedsigner #507: Docs: Verifying Software -- Old mac versions of shasum don't support '--ignore-missing' flag

jdlcdl, 2023-11-09:
As reported in the telegram group by Guy:

Some older mac users may have issues verifying software downloads because their version of `shasum` does not support the `--ignore-missing` flag.   Guy was able to complete verification simply by leaving that flag off the commandline.  Later, they confirmed that this is a known issue:

[image]

I suggest that a note similar to above is included in our installation docs so that users will know to avoid this error.

SeedSigner, 2023-11-22:
ACK

Chaitanya-Keyal, 2025-04-02:
Hey @jdlcdl, @SeedSigner,

I've linked a PR that resolves this issue. Please do review it and let me know if it requires any changes.
Thanks!