SeedSigner/seedsigner #508: Add "Calculating" spinner when creating new seed via image entropy

kdmukai, 2023-11-17:
Steps to reproduce:

* Start new image entropy seed process
* Click joystick to take final pic
* Joystick right to accept pic
* Select 12 or 24 words
* ...slight delay here...

It's just enough of a delay that I keep wondering if my click registered.

## Improvement
Use the loading spinner thread to provide immediate UI feedback during the ~750ms it takes to create the seed.

SeedSigner, 2023-11-18:
I've noticed this too, agree that this enhances the user's experience. 🙂

krishnavbajoria02, 2025-04-03:
Hi @kdmukai and @SeedSigner,

I’d like to propose a fix for this issue. Here is my proposed approach:

1)Introduce a loading screen ```(LoadingScreen(_("Calculating...")))``` right before entropy hashing begins.
2)Use threading to run the entropy processing in a separate background thread, ensuring the UI remains responsive.
3)Stop the spinner once entropy hashing is complete, then proceed to the next step.

Below is the updated run() method of ```class ToolsImageEntropyMnemonicLengthView``` implementing these changes:
```
import threading
from seedsigner.gui.screens.loading_screen import LoadingScreen

#Modified ToolsImageEntropyMnemonicLengthView.run() method
def run(self):
    button_data = [self.TWELVE_WORDS, self.TWENTYFOUR_WORDS]

    selected_menu_num = ButtonListScreen(
        title=_("Mnemonic Length?"),
        button_data=button_data,
    ).display()

    if selected_menu_num == RET_CODE__BACK_BUTTON:
        return Destination(BackStackView)

    mnemonic_length = button_data[selected_menu_num].return_data

    # Show "Calculating..." spinner
    loading_screen = LoadingScreen(_("Calculating..."))
    loading_screen.start()  # Starts the spinner in a separate thread

    def generate_seed():
        try:
            preview_images = self.controller.image_entropy_preview_frames
            seed_entropy_image = self.controller.image_entropy_final_image

            # Incorporate CPU serial number entropy
            try:
                stream = os.popen("cat /proc/cpuinfo | grep Serial")
                output = stream.read()
                serial_num = output.split(":")[-1].strip().encode('utf-8')
                serial_hash = hashlib.sha256(serial_num)
                hash_bytes = serial_hash.digest()
            except Exception as e:
                logger.info(repr(e), exc_info=True)
                hash_bytes = b'0'

            # Add timestamp-based entropy
            millis_hash = hashlib.sha256(hash_bytes + str(time.time()).encode('utf-8'))
            hash_bytes = millis_hash.digest()

            # Add entropy from preview images
            for frame in preview_images:
                img_hash = hashlib.sha256(hash_bytes + frame.tobytes())
                hash_bytes = img_hash.digest()

            # Add entropy from final high-res image
            final_hash = hashlib.sha256(hash_bytes + seed_entropy_image.tobytes()).digest()

            if mnemonic_length == 12:
                final_hash = final_hash[:16]  # Use first 128 bits for 12-word mnemonic

            # Generate mnemonic
            mnemonic = mnemonic_generation.generate_mnemonic_from_bytes(final_hash)

            # Clear entropy data from memory
            self.controller.image_entropy_preview_frames = None
            self.controller.image_entropy_final_image = None

            # Store the generated seed
            seed = Seed(mnemonic, wordlist_language_code=self.settings.get_value(SettingsConstants.SETTING__WORDLIST_LANGUAGE))
            self.controller.storage.set_pending_seed(seed)

        finally:
            loading_screen.stop()  # Stop the spinner when processing is done

        # Proceed to the next view
        return Destination(SeedWordsWarningView, view_args={"seed_num": None}, clear_history=True)

    # Run seed generation in a separate thread to keep UI responsive
    threading.Thread(target=generate_seed).start()

    return None  # Let the spinner continue until the thread finishes
```
I would appreciate any thoughts on this approach! Please let me know if there are any suggested refinements or alternative ideas. If this looks good, I can proceed with opening a PR.
Thanks!

kdmukai, 2025-04-04:
Also note that the delay in the seed generation is probably due to the live preview image buffer being completely full of images that need to be hashed into the final result.

So the delay might be barely noticeable if there are very few preview frames (user clicks for the final pic very quickly), up to some (presumed) worst-case scenario where the buffer is hashing the max number of preview frames (user has the live preview running for a while before clicking for the final pic).

Also note that the buffer is a rotating window of the N most recent frames, so the buffer will not grow indefinitely.

krishnavbajoria02, 2025-04-05:
@kdmukai 
Thank you for more information regarding this issue. I will look into it and then proceed with opening the PR.

krishnavbajoria02, 2025-04-05:
@kdmukai 
I checked for similar implementations and found this kind of implementation : 
https://github.com/SeedSigner/seedsigner/blob/a3c8f390eb133d8302ba10605da5d003ae046e74/src/seedsigner/views/psbt_views.py#L85-L106
In my PR I'll try to follow this implementation pattern only then if my approach was ok [here](https://github.com/SeedSigner/seedsigner/issues/508#issuecomment-2775227341).

PR #711 (2025-07-04): [Feature] Added 'Calculating...' spinner while generating seed from Image Entropy
## Description
Fixes #508 
This PR adds a "Calculating..." spinner during the image entropy seed generation process. The spinner provides immediate feedback to users during the delay caused by hashing preview frames and generating the final entropy-based mnemonic seed.

The delay occurs because:
1) The live preview image buffer contains multiple frames that need to be hashed into the final result.
2) The hashing process depends on how long the user keeps the live preview running before capturing the final image:
 [... cut]