SeedSigner/seedsigner #510: Behavior of back button confusing when entering password

SeedSigner, 2023-11-20:
I received this feedback from a user, just documenting here for consideration:

"I was entering a long passphrase, and twice accidentally the joystick moved up to the back button, pressed, and the whole passphrase was deleted before I could finish entering it. If possible in future updates, make the back button more segregated from the keyboard so that you have to be intentional about highlighting it?"

Richie-BTC-Rich, 2024-03-18:
Agree, I have this issue a lot.  The press down to enter on the joystick is sketchy.  Sometimes it moves the selection and enters the wrong selection. other times if on the top row it will move up on the press down, pressing the "<" at the top and exit the whole process.  In my opinion entering via a press down on any screen is troublesome, would be much better to move with joystick and enter with buttons on right.  For example on the passphrase screen the buttons on the right could be Top - toggle through changing case, special chars. Middle - Enter , and Bottom - Check or "Done"  .  I will say that I'm fairly certain this is due to the "softness" of the joystick cap knob if you have one on your joystick nub. The less rigid the knob is makes the press down not as effective. ( imagine a cotton button vs metal one). It seems a proper more rigid joystick knob would help some.

kdmukai, 2024-03-20:
That's good feedback.

[image]

We're obviously super-constrained by the available screen real estate.

I like the idea of using one of the right buttons as the SELECT action for each char input. But it would be hard to communicate that visually (maybe the virtual button next to it could display the currently selected character?). And we'd still have to allow joystick center click as SELECT (otherwise we'd break the existing UI/UX expectations) but obv advanced / wise users could just avoid using it.

Note that the top right button's cycling would have to be:
* uppercase
* lowercase
* digits
* symbols 1
* symbols 2

It's not horrible to have to cycle through 5 keyboard layouts, but it is a slight tradeoff.

---

Another option would be to try to fit in a selectable "DONE" key (e.g. next to "del") to free up the bottom right hardware button. I think I'd prefer this if it can fit without overly crowding the bottom. We can make the spacebar a bit skinnier to create a little more room (just noticed that its width doesn't align with the keys above it. Ug).

kdmukai, 2024-03-20:
In the meantime, if the passphrase isn't blank when you hit the "<" BACK arrow, we could throw up a "Are you sure you want to cancel?" dialog to at least prevent the worst-case scenario of accidentally losing all the work to type in a long passphrase.

Richie-BTC-Rich, 2024-04-12:
Yea I think the addition of a [Done] at the lower right selectable by the joystick would be perfect as well as the exit warning on selecting the "<".  

Initially I was confused the check mark was the enter or select button for the selection on the left .  To me I'm confirming the selection on the left with a "check" that's the one I want opposed to the check meaning I'm done entering the passphrase, which by adding the [Done] selection to the right of the DEL would take care of.

Top Button switching alpha case
Middle Button switching digits, special 1 and special 2 
Bottom Button either leave it a check mark or have it update with the selected character on the left would work

Im trying to remember how the Xbox 360 and Playstation3 handled this with the controller , but I don't have those anymore..  I know it was intuitive but they had a lot more screen to work with too.

closed 2024-07-14 by newtonick