SeedSigner/seedsigner #515: Keypad PIN code securing QR codes idea

Runixcube, 2023-12-27:
here’s an idea that I need peer reviewed please. It appears sound to me but I could just be engineering backwards or something. I wouldn’t know without others reviewing it. Thanks. 

[Keypad Seed Word security idea.pdf](https://github.com/SeedSigner/seedsigner/files/13780194/Keypad.Seed.Word.security.idea.pdf)

jdlcdl, 2023-12-28:
Just some thoughts:

There is a likely non-zero probability that words in a mnemonic are duplicated, so perhaps some thought should be given to duplicates in the grid of "random words" which are NOT part of the mnemonic... for hiding the instances where the scheme suggests that duplicates exist in the same pin-code button box.

This entire scheme, as well as all of the variations are what I like to call "meta secrets".  Combined, they're all part of your "secret" and I mean that in the context of "Can you keep a secret?" which is a question we all must ask ourselves... and we must live with the consequences of the reality far into the future as opposed to how sincerely we answer that in the present.  Complexity, for the most part, works against us... so does our own ability to remember secrets and meta secrets, and also how we plan to transfer those secrets to our heirs.

This scheme feels much like "border wallets", except that my gut-feeling is that it's slightly less secure (by some factor of four but I'm not sure how that plays out exactly) and maybe even more if the expanded grid does NOT cover all 2048 seed words while also being slightly easier to remember/execute... but the scheme itself is also a secret, especially if few are using it.

A strong bip39 passphrase can be as 'hard' (or even harder) than a mnemonic and is a single additional "secret" that can be easily be transferred to an heir, and/or recorded in case we forget.  Also, it provides for "decoy" wallets.  It has been vetted by many, and even if your heirs only had the following: "My funds are hidden in mnemonic X with passphrase Y and I used standard wallets when I stacked during the first 4 epochs." then they'd have a very good chance of learning that they must keep their secrets, that you're referring to bip39, and that native-segwit derivation was likely used.  Assuming they knew nothing more than "don't trust, verify", they'd easily find the help they need to recover your treasure.  Personally, during my last breath, I want to be able to relax in peace and enjoy that moment, instead of thinking "oh no! what have I done?".

Besides all that... I really don't have any major issues with this... we all have to weigh our own threat-model and keep our secrets secret.

Thank you for writing this up, for your deep thoughts, for your creativity, and for being here to share your ideas with others.

p.s. A shameless plug to my recent-ish "Can you keep a secret?" thoughts is [here](https://primal.net/e/note1pqju4lcr9ud2nzrlsnq2sr7nrdf54aeecwq7t0mpmsu2832juehsf3ztfc)... a crowd which you, apparently, are a part of too! ;)

Runixcube, 2023-12-28:
Thanks for taking time to read my work and then commenting. 🙏 

Yes it would need someone smarter than me to do the math to say how secure it is though it is certainly more secure than seed words on a plate or paper that is common at the moment. 

I read the piece you wrote/ linked too. Thanks, it was very thought provoking. 

Have a nice day

jdlcdl, 2023-12-28:
Wondering if you can provide an example, because I'd like to try to break it... but I'm not interested in trying to crack it if it has a bip39 passphrase.

Make a 12 or 24 word mnemonic, record it, and the fingerprint.
Make an example of your grid w/ 4 words per cell, in whatever rotations you like... but in a shape where a pincode overlay would cover all your words within the standard 1-9+0 buttons.

I will do my best.

Runixcube, 2023-12-29:
Excellent idea. I should have included one in my original write up, I’ll update it (and my first post as I see the software here does all the formatting)

Bellow is my idea in its most basic form, grid lines are intact and the two bottom areas are blank, I could have filled them and taken the lines away but seeing as this is merely a test, proof of concept and you know what it is there’s no point.

Everyone likes a good mystery, so…

Clearing out someone’s belongings after death you find this piece of paper. No other clues just this. 

[image]

jdlcdl, 2023-12-29:
I think I'm starting to understand some of the math involved in brute forcing this.

Assuming that: 
* no rotations are done, all four words are as is (not scrambled) and none are ignored...
* and no repeated cells in the pin (otherwise duplicates of 4 adjacent words "feels" unlikely):

...brute forcing different pins would require:
* 720 attempts for 12 word mnemonics: because 10 * 9 * 8
* 115200 attempts for 24 word mnemonics: because 10 * 9 * 8 * 7 * 6 * 5

In a 12 word mnemonic, the 12th word contains 7 entropy bits (and 4 checksum bits), so out of the 720 attempts, roughly 128/2048 or 6.25% of those will be valid mnemonics and the rest of the brute force attempts can be ruled out.

In a 24 word mnemonic, the 24th word contains 3 entropy bits (and 8 checksum bits), so out of the 115200 attempts, roughly 8/2048 or 0.39% of those will be valid mnemonics and the rest of the brute force attempts can be ruled out.

On my desktop computer, this takes 1m13s for the example provided above, and reveals 50 valid 12word mnemonics and 577 24word mnemonics that are valid, which is "ballpark" for the above estimates.   BUT:
* Would still need to seed a bip32 master wallet, derive some standard paths, and check the blockchain to see if funds are there.
* And that's with the above assumption that no rotations have been employed and that all-four words per cell are used... that none have been scrambled, and none have been ignored.  It may quickly get out-of-reach for me if any of those assumptions are broken (while it may be out-of-reach for a user's brain memory too, leading to loss).

Would you consider putting some funds into a UTXO within this wallet on testnet?... or providing an address from that wallet, in a response below, so that we can pretend that it's trivial to scan the blockchain for funds?  I will then try to reveal your pin and mnemonic.

Runixcube, 2023-12-29:
Thing is if it’s as easy as you’re suggesting the funds will be gone before you find the wallet as others maybe watching.

if all you found was the paper could you find the wallet with no further details?

Runixcube, 2023-12-29:
Earlier I looked at the linked page bellow and also asked them to take a look at this to see what they think. 

[https://www.whatisbitcoin.com/security/guess-my-seed-phrase](url)

Runixcube, 2023-12-29:
I think I should give more details of what I’ve done so we can work this out faster and not waste time as we’d need to explore this anyway and you can do the math easier. 

There is rotation or reordering. How many I won’t say. 

There may or may not be decoy wallet or wallets.

jdlcdl, 2023-12-29:
> I think I should give more details of what I’ve done so we can work this out faster and not waste time as we’d need to explore this anyway and you can do the math easier.
> 
> There is rotation or reordering. How many I won’t say.
> 
> There may or may not be decoy wallet or wallets.

If there are rotations, and an arbitrary amount of them, or if there are ignored words so that maybe it's more than 3 digit pin for 12 or 6 digit pin for 24... then it quickly gets much harder than mine.  I cannot know your pin or mnemonic without knowing at least an address to search for, or a pubkey, or a fingerprint.  Knowing those gets us the "must search the blockchain" task from trivial-but-resource-and-time-consuming to immediate... but it's hardly so difficult that it could not be done by a determined attacker who knows there is something worth-while to steal.

My computer found many possible pins/mnemonics.  I could give you a list and the code...but I'd love to have something to search for just to take the mystery/experiment as far as we can (while it's fresh in my head).  I wouldn't share the code without your confirmation that there are no real funds on this... and that you haven't put real funds behind this method and also lost that piece of paper yet.  I'll share as soon as we've exhausted the task and you confirm that it's safe to do so.

Btw: 2 more cells makes it harder but only trivially so (12 * 11 * 10 instead of 10 * 9 * 8).  If it were a large sheet of cells and you had to lay a virtual keypad in the right position, and rotations were mandatory, it might get out of reach and into computationally-hard territory, which is the real goal.  If it's a good idea, then it's a good idea for many to do... and then it becomes a good idea for attackers to optimize code for cracking real funds in the wild.

Runixcube, 2023-12-29:
If im understanding - 

- it’s too time consuming to try all the seed words you’ve found without funds in the wallet?

- to speed it up you need me to deposit funds in the wallet?

- You would then have to run every seed phrase and see where the funds are?

- but you don’t want to do that as it’s too time consuming but not out of the scope of someone else

- so you need some public address?

is the public info you need just to speed up the process as the article I linked to above all they shared was the seed words and say finders keepers.

 Can’t I just do that as time it takes to disappear is important here as if someone used this method whilst still alive time could be what stops them losing their bitcoin.

jdlcdl, 2023-12-29:
I have not figured out the code for 1) rotated words in one or more of the cells nor for 2) scrambled words in one or more of the cells, but I'm thinking about how to implement that currently (so far, Im mind-boggled) and also just trying to figure out how to estimate the math as I did above.

With the unrotated assumption, I've already found 50 12 word seeds that are valid and 577 24 word seeds.  So they're all valid wallets -- without bip39 passphrases, and I'd need a target to shoot for.  The fingerprint would widdle it down bigtime, likely I'd land exactly on it for each match, with only 1 in 4-billion chance of error.  Else, the very first address on a known derivation path is almost as easy (but does include a few more sha512 hash rounds).  With only an address... and if you tell me it's one of the first 10 receive addresses on a standard derivation path... It would be 10x harder but every small non-standard change we make to that target makes it harder.

Since you've mentioned that there are rotations, we can already assume that I have not yet found it... but I should be "capable" to find it once I figure out the implementation of doing so... but also Im not sure if it will be computationally out of reach since I haven't figured on how to calculate the math of the brute force.

True that not having a target or having the target harder to find could keep funds safe.  But we have to assume mistakes by the user and perfection by the attackers.  They already have an instantly fast index of all addresses that hold funds on the blockchain (like almost exists in blockchain explorers), and they're only attacking you because they know you have more funds than the attack costs and they already found your backup like above.

Runixcube, 2023-12-29:
One other thing I forgot to mention. We need to explore this to see how viable it is. 

- There is a double repeating digit/number in the sequence.

jdlcdl, 2023-12-29:
So are you saying that there are 4 duplicated words?

jdlcdl, 2023-12-29:
About exploring this as a backup method.  If it is easy to crack without duplicates but impossible to crack with duplicates, then requiring duplicates becomes a rule, not an option, else someone who chooses not to duplicate (which is not even hardly likely in a real mnemonic) won't be protected.  Same with rotations, if adding one or more rotations makes it secure and not having any rotations makes it insecure (I'm getting the feeling that this is the case), the it become mandatory and not optional, else someone using it who does not rotate (and take the risk that they'll forget what the rotation was) is not protected.

Runixcube, 2023-12-29:
No duplicates in the seed phrase. I just gave away that it’s a 12 word seed. 

Also I just thought 

- can also use a 4 digit passcode which would use 4 squares and ignore 1 word from each. Add in reordering or rotation to the mix.

jdlcdl, 2023-12-29:
Im misunderstanding

> There is atleast one double repeating digit/number in the sequence.

If that statement is true, I'm thinking that it means the pin might read like 133, and then there would be 4 duplicated words.  What did you mean by "there is at least one double repeating digit in the sequence."?

jdlcdl, 2023-12-29:
we're quickly getting into the very realistic/probable realm of meta-secrets that would have to be recorded on the sheet, else they'd be forgotten and funds lost.  Even with just some scrambling (because the mnemonic is sliced up in pin-digits), I think it's likely a good idea that the user would write the wallet fingerprint on the sheet... so that they know when they get the right pin... but maybe not.

Runixcube, 2023-12-29:
For a 12 word seed I’d just write a pin like this and/or remember it

Example 

- 12<ins>3</ins>436

Means 3rd digit is read bottom two words not top. Then next number 3 is read top as the bottom has already been used. 

I agree it could start getting difficult to remember and that’s something to avoid so what is the bare minimum that can be gotten away with?

jdlcdl, 2023-12-29:
Originally, my thought that this was like a border wallet, except easier by a factor of 4... because it can be assumed that with each cell of four words, the words are constrained to being adjacent to each other (even if they are rotated or worse, out of order).

But it becomes much harder to crack once it's optional to reorganize the words and even take them out of their boxes.  It becomes much harder than 24 scrambled seed words, because it could be any 12 or 24 of the 40 currently, or any 12 or 24 of 48 if you use the other two boxes on the bottom.

As I mentioned earlier.  I think I can provide a list that includes your mnemonic IF AND ONLY IF, it was done without rotations, ignoring words, and scrambling.  If that is an optional rule that users might employ... then this method seems trivial... but if it's mandatory to rotate/scramble/ignore some... then my gut feeling is that it can become computationally hard to crack... while also being hard to remember for the user.

Runixcube, 2023-12-29:
Words are never removed from their box.

Runixcube, 2023-12-29:
> Im misunderstanding
> 
> > There is atleast one double repeating digit/number in the sequence.
> 
> If that statement is true, I'm thinking that it means the pin might read like 133, and then there would be 4 duplicated words. What did you mean by "there is at least one double repeating digit in the sequence."?
> 
> That said, while it's not likely, it is possible... and it makes the computation a little bit harder.

Sorry I missed this earlier. 

there are no duplicated words in the seed. 

The seed is 12 words

a repeating digit means like this example

112345

This would mean the top could be read on the first pass, the bottom words on the next 

ie

112345

would be

cheap miracle use daughter glare empower other orchard shoot merit loyal gap

that is without rotation. 

Say it was 

<ins>1</ins>12345

Now becomes 

use daughter cheap miracle glare empower other orchard shoot merit loyal gap

because the bottom was read first on the first digit.

jdlcdl, 2023-12-29:
So the 1's being adjacent in the above example, is it correct that the pin code can also be "12345", with no step of having to remember to ignore? and instead having to remember to start at the bottom or start at the top with rotations?

If the 1's can be separated, as in 121345, then I can see it, but this is what I meant by stripping words out of their boxes (ignoring is sort of doing that too).

All of these possible meta-secrets and permutations makes the code-cracking MUCH more difficult... likely into "good enough" as security goes... but I still think it either requires the user to record more info on the paper... or they risk losing funds.  And if all of this is optional, then we should assume the user will do what is easiest, and not rotate anything, not ignore anything... since it's within the rules.  Our assumptions should be "The user will be sloppy, forgetful, lazy and that attacker is the smartest guru that we'll never even get a chance to meet or become."  We should be left with simple to employ for the user... and not enough resources in the universe for the attacker to have a good chance of winning.

btw: Im still thinking about how to implement the rotating of words efficiently.

jdlcdl, 2023-12-29:
I wish I'd paid attention in math classes... (and that if I did I'd remember them still).

My gut feeling is that *arbitrary* rotations/permutations of words in a box means 4*3*2*1=24 for each pin digit, starting with a 12 word seed, this would be 10*24 * 9*24 * 8*24 = 9953280 possible 12 word combos with 622080 of them likely to be valid mnemonics... less hard than scrambling 12 words (already a bad idea).  For a 24 word seed it becomes 10*24 * 9*24 * 8*24 * 7*24 * 6*24 * 5*24 = 28894769971200 with 112870195200 of them likely to be valid mnemonics.  I'm not convinced that 100s of trillions is quite into big-number territory but I'm sure it's more than Im willing to run on my scrawny 4core xeon desktop... in python.  Still, it should be easier than cracking 24 scrambled words.

I'm not absolutely sure about those numbers there, but it leads me to believe that I should gracefully back-away from pursuing trying to crack your seedphrase given that arbitrary rotations and permutations are a possibility.

Thank you for this challenge, it's been fun, and a useful exercise... but I'd worry about loss of those meta secrets and I wouldn't rule out a determined attacker with resources and reason to challenge this method.

Runixcube, 2023-12-29:
I don’t see the number repeating as ignoring or jumping boxes. I just see a pin pad with a PIN code.

so I’m just inputting a pin. Easy enough to remember or hide else where. 

As for the rotation I would use one method and remember that ie first pin is reversed, last pin is reversed. 

I also should have paid more attention in maths. 

It’s been good fun working through this with you and I do agree that I have started to feel it maybe easier to crack than 24 scrambled  words but thought the extra words upto 48 might level things out.  

I still wondering if you have found the seed and/or just the 2 decoy seeds I placed.

Runixcube, 2023-12-29:
> So the 1's being adjacent in the above example, is it correct that the pin code can also be "12345", with no step of having to remember to ignore? and instead having to remember to start at the [...]

Pin can be anything that helps you remember where you put your seed. If you haven’t had a go at making your own example maybe have a try. When I was making it I remembered that you said about hash numbers and I went reading. I saw that the last number is a hash so I made a point of including decoy wallets rather than just random bip words in the hope it would throw in even more wallets for an attacker to trawl through thus increasing their time, energy and money. 

For a 12 word seed 123456 would just mean the first two words in each box as 6 * 2 = 12

It’s not hard for me to remember a pin and that i  should read the first digit backwards and or last or even every other digit. 

 For a 24 word seed with 6 digit pin say 123456 all words would need to be used per box as 6 * 4 = 24. 

Rotation is what saves it as it increases the randomness as does the words being in different boxes but when you run a computer program you’re just putting all the possible seeds together and you’re not even trying to work the pin out.  

With not everyone  having a computer program to do that calculation it’s safer than a list from a find and enter type attack by someone in the house, disgruntled spouse, child etc. A simple 6 digit pin even without rotation is probably good enough for that situation?

jdlcdl, 2023-12-29:
> I still wondering if you have found the seed and/or just the 2 decoy seeds I placed.

I only implemented the non-ignoring/non-rotating version.  It found 50 12word seeds and 577 24word seeds, but if you did any scrambling then none found would be yours.   While I did the math in the previous post, I never implemented code that would exhaust those arbitrary permutations per cell.

Runixcube, 2023-12-29:
In all this I almost forgot. QR codes are never scrambled. This method achieves that. I wonder if you feel it has any value for that application?

Runixcube, 2023-12-29:
I thought I’d share this as I believed it could be of use, therefore it was important that it be tested by others. I had contemplated that someone smart would take one look at it and laugh saying it’s a basic math problem and easily solved/cracked.

Just storing seed phrase amongst other words is enough for most people alongside a passphrase. As seedsigner said the other day “words are everywhere”

I do have further ideas on this, a physical object, but I think it might get complicated and then like you say people lose their funds. 

Have a nice day and again thanks for your time 😊