SeedSigner/seedsigner #520: [Minor Improvement] Re-word warning screen title when multisig change/receive addr fails verification

kdmukai, 2024-01-10:
Just noticed that the single sig verification failure...

[image]

...is more strongly worded than the multisig version:

[image]

---

I recommend a trivial change in the multisig side from "Caution" to "Suspicious PSBT".

There's explicit logic to opt for the softer "Caution" in `PSBTAddressVerificationFailedView` but I can't think of any reason nor recall any rationale for this.

If there's nothing deeper to consider here, this would be a simple, easy first issue for a new contributor.

krishnavbajoria02, 2025-04-03:
@kdmukai 
I think this issue has been fixed as per the code in the dev branch. 
https://github.com/SeedSigner/seedsigner/blob/a3c8f390eb133d8302ba10605da5d003ae046e74/src/seedsigner/views/psbt_views.py#L454-L479
But if you think anything else is remaining, please do let me know. Thanks.

Bicaru20, 2025-04-03:
@kdmukai 
I've just tested it in the seedsigner [emulator](https://github.com/enteropositivo/seedsigner-emulator) and as @krishnavbajoria02 mentioned, the change has already been made:

[image]

If no other actions are required, maybe this issue could be closed.

Regards.-