SeedSigner/seedsigner #527: Loaded Multisig Descriptors are wiped at Main Menu

3rdIteration, 2024-02-17:
**Expected Behavior**
Once loaded (either through scanning a descriptor from the main menu or loading one through the address exporer) a multi-sig descriptor is automatically used to verify multisig transaction outputs. 

**Actual Behavior**
Multisig Outputs are not automatically verified unless the descriptor is loaded each time a transaction is signed.

**Exploration**
As it stands, the current multisig descriptor is wiped each time the user returns to the main menu.

newtonick, 2024-02-21:
The behavior of forgetting previously scan descriptors is intentional. This said I'm personally open to changing this behavior to store descriptors in a single session (during power on state) if done in an intuitive way. I like your idea of changing "Next" to "Skip Verification" in the UI.

3rdIteration, 2024-02-21:
Yea I'll fix the PR I brought to just tweak the language and add a warning. 

How exactly it behaves is less important than making sure that it's both consistent and obvious what is going on.

3rdIteration, 2024-02-21:
Ah actually I see what happened there, I merged the dev branch, so I have just wiped those commits, as well as revert the change to keep it and the PR is back to the warning only now.

New PR is here: https://github.com/SeedSigner/seedsigner/pull/533

Whether you think the dire warning is appropriate is neither here nor there, but this is a situation that can result in loss of funds, so the use of a "Dire Warning" here seems appropriate.