SeedSigner/seedsigner #531: [Feature Request] Fullscreen SeedQR export

BradleyAllanDavis, 2024-02-20:
Currently, when exporting a seed phrase via QR code, the two options given (Standard and Compact) **do not display the QR code using the entire screen of the device**. This makes it difficult to scan using phones, hardware wallets (including SeedSigner), and backup devices like SeedHammer. An option to display the SeedQR using the whole display like is done when exporting xpubs would make it much easier to export the seed.

newtonick, 2024-02-24:
[image]

Just to clarify, the above image is not large enough and you want the SeedQR to display from edge to edge?

BradleyAllanDavis, 2024-04-13:
@newtonick Correct. I am not able to import this QR into other HWWs and other wallet software.