SeedSigner/seedsigner #536: Backup source for seedsigner gpg keys

vnprc, 2024-03-15:
When I try to follow the instructions found [here](https://github.com/SeedSigner/seedsigner#verifying-that-the-downloaded-files-are-authentic-optional-but-highly-recommended) to verify the seedsigner img file I get this error attempting to download the gpg keys:
```
$ gpg --fetch-keys https://keybase.io/seedsigner/pgp_keys.asc
gpg: requesting key from 'https://keybase.io/seedsigner/pgp_keys.asc'
gpg: WARNING: unable to fetch URI https://keybase.io/seedsigner/pgp_keys.asc: Server indicated a failure
```

Is there an alternative source to retrieve `pgp_keys.asc`?

jdlcdl, 2024-03-18:
The above worked for me but I'm trying it 3 days after the above error was reported.  Can also verify...
* in main repo: https://github.com/SeedSigner/seedsigner/blob/dev/seedsigner_pubkey.gpg
* as a key fingerprint on twitter: https://twitter.com/SeedSigner
* in person (someday hopefully) as he turns his head in response to "zero one one nine?  zero one nine-teen?" ;)

Marc-Gee, 2024-03-19:
VNPRC, Thanks for opening the issue. Currently there are 3 online, independent locations where the projects Pubkey is stored. Our Website , Twitter and GitHub, and we rely on Keybase to cross-check those 3. 

If Keybase fails temporarily then the user should import it from the GitHub link per JDL answer above, and then perform additional verification steps that it's the _correct_ pubkey when running the subsequent GPG verify command. 
 

Keybase.io remains, ( in March 2024 as I write this)  the only service I'm aware of that can: validate that the public key as published to Keybase is the same as the 3 other independent locations. It is validated by Keybase service periodically, and also by the users when interacting with Keybase (ie the blue proof badges). 

That said, if Keybase does end it's service (or temporarily not function as you discovered),  we need to ask users perform some additional, manual steps to :  import the pubkey from the GitHub link above, by modifying the gpg command to point to the file on GitHub; and then subsequently verify that the pubkey which was imported, is in fact in agreement with another 1 or 2 locations;  and that the verify command is returning the correct public key also. 

The key locations that we can direct users to in that case are the GitHub  for the ASC file import and then the website seedsigner.com and the projects Twitter Pubkey post (previously validated). https://twitter.com/SeedSigner/status/1530555252373704707 for the confirmation of the verify command. 

Without Keybase's functioning, our project (and also other Bitcoin projects) will need to have their users perform those extra steps to manually cross-check the public key they imported into their GPG keychain and also the public key value returned when running the GPG verify commands (if they get a 'good signature ' Result.

If the OP or anyone else is aware of websites/services to safely improve or simplify the software validation process, please let me know. 

The new 1.84 verification command in sparrow wallet seems a possibility, but I have not dug into it yet, and also any dependence /requirements it might place on the Seedsigner project. 

Thanks again for highlighting this issue now, which might become a permanent issue if Keybase becomes unreliable in the future. It is the first known failure report in about 2 years. 

Happy to discuss further. 
Marc.