SeedSigner/seedsigner #542: [Feature Request] Restore using BIP-39 Numbers(Optionally display number during backup)

CODLCO, 2024-04-10:
We have found that we can save a significant amount of space physically by recording the bip-39 underlying number vs the word its mapped to and would like to request a restore by number option be made available. 

It was my understanding that the entropy for a seed(binary) is chunked into 11 pieces(thinking 12 words only here) and then each piece is converted into decimal and then mapped to a word on the bip-39 word list. 

Allowing users the option to directly input the decimal chunks for their associated word would help reduce their backup and restore time for products that allow users to store their seeds as these numbers instead. 

Optionally we could add a toggle or option to see the numbers that were used to map the associated words enabling the Seedsigner to allow users of certain products to benefit from a faster backup option for certain products. It would also add transparency to the backup process.

Concerns over error checking can be mitigated by double checking and test restoring before relying on the backup and moving any funds.

CODLCO, 2024-04-10:
Displaying the number and restoring using the human readable list found here:
https://github.com/bitcoin/bips/blob/master/bip-0039/english.txt

emanuelet123, 2024-08-13:
I think it would be a really nice option to have, because there's people that only keep the bip39 numbers in metal backups. And I guess the 1-based should resolve the problem, 1-based is more for user friendly, words from 1 to 2048. Looking foward to have this feature in the next version!

jdlcdl, 2024-08-13:
Just adding a note here that a similar issue already exists, with relevant commentary.
https://github.com/SeedSigner/seedsigner/issues/211

nomadbit21, 2026-10-01:
Opened a PR that shows the BIP-39 index (1–2048) and its 12-bit row during seed backup, with the existing reveal warning and a backup test.

It also covers the load side requested in #211. Not Stackbit 1248.

https://github.com/SeedSigner/seedsigner/pull/1051