SeedSigner/seedsigner #544: Feature Request - Add Signet as Additional available "Bitcoin Network"

FractalEncrypt, 2024-04-15:
I use Seedsigner all the time to teach people about using Bitcoin. I am able to do this free an easily using Testnet. Well, I was, up until now. It appears that degens have found testnet and aside from it's well-known and understood issues, it is now difficult for users like me to send transactions due to network congestion. 

So I think it may be time to consider adding functionality for users to select the Signet network for things like teaching people. Regtest is great if you're just developing on your own, and testnet has been a solid and wonderful resource as well. 

So the request is simple;
(I say this understanding the implementation may not be so simple)
In addition to having the 3 current options
Bitcoin Network
Testnet
Regtest

we add functionality for a 4th network option: Signet.

jdlcdl, 2024-04-15:
Until others get a chance to look this over,

I'm only aware, under the hood, that seedsigner will use a different key into the embit.networks.NETWORKS dictionary which helps with versions/magic-bytes/etc that change xpubs/addresses/etc to forms that will be refused if created for one network and submitted on another.

However, besides the "name" key "Testnet", "Regtest" and "Signet" all other key/values appear the same between these 3 networks, only "Mainnet" has unique values. In theory, if you setup for regtest or testnet and point your coordinator and node to signet, all should still work from seedsigner's perspective.

I have not actually tried, if you have problems, I appologize in advance and will look forward to learning where this isn't the case, and then searching for a fix.

FractalEncrypt, 2024-04-15:
Thank you this is amazing. I was able to use Sparrow on Signet with the Seedsigner set to Testnet as you suggested. I could do everything fine as far as a regular export of xpub to Sparrow and then signing of the PSBT in Seedsigner, sending the signed tx back to Sparrow to broadcast. Transaction successfully confirmed.

jdlcdl, 2024-04-15:
And if I don't believe you? can I ask for verification?  

I'll send 2:1 for each signet coin you send here:
tb1qn466ntln6gsgl4076neg2n59sxzgkjp2nmh8e0

...but you may have to wait for my signet node to catch up.
...and you'll have to give me a chance to get some coins from a faucet.
Thanks for asking about this.

FractalEncrypt, 2024-04-15:
I will do that! And also here's the TxID from my post above - 7a175a1297894ca56655c24b937d1740f282d94a48b97790a39435e6d0a6e796

And screenshot from the Sparrow UI with the tx details.

[image]
I just sent you signet sats via Seedsigner/Sparrow PSBT - txID = 8c582386388df3c3921183d68ab7b41c6e16957ab0b92b5f09c205dcdf20a574

FractalEncrypt, 2024-04-16:
Rec'd! So it worked for you too, no problem?

jdlcdl, 2024-04-16:
> Rec'd! So it worked for you too, no problem?

Seedsigner made no problems for me last night.

I had problems but they were on my side (configuration, and existing problems on my computer yet to be resolved).