SeedSigner/seedsigner #546: [Enhancement] MicroSD Toast Timer

easyuxd, 2024-04-26:
## Proposed Enhancements:

### 1. Add automatic dismissal of MicroSD Toast message.
- Recommended timer: 5 seconds.

**Rationale:**
- The current toast implementation adds friction by obfuscating a portion of the UI, rendering the "Tools" and "Settings" labels unreadable until a button is pressed. 
- New users may not be aware of their menu options unless they first dismiss the message.
- Many product photos of the v0.7.0 are now inadvertently showcasing an informational message rather than SeedSigner functionality.
- By adding a timer, users that rely on this message continue to get a reminder on boot

### 2. Add advanced setting to disable toast

**Rationale:**
- Some users may not need the reminder or do not want to remove the MicroSD.

---

## Proposed Design

**5s timer:**

[image]

alvroble, 2024-07-17:
Looking into RemoveSDCardToastManagerThread, wouldn't this be easily achievable by modifying the supperclass init call?

https://github.com/SeedSigner/seedsigner/blob/0f329f6fb568f1c7aaae6288ce33e38c566f3d1d/src/seedsigner/gui/toast.py#L189-L196

Setting `duration=5` would provide the requested behavior. I tested it with this change in Rapberry Pi OS build and it worked as expected. A question for other devs: do you think this change could create any side effects?

jdlcdl, 2024-07-18:
In the lead-up to v0.7.0, this was such a big deal (that user could remove any capability for secrets to be stored... by a software that is already super-paranoid about never storing any secrets in the first place) that many (including me) advocated a scary warning screen to entice the user to remove the sdcard.

The only way that secrets would get written, is if the evil maid got to the sdcard, and then removing it wouldn't be helpful because she'd just have you sign every tx to her own wallet... whether or not the sdcard was removed.

Back to your question, I believe that your proposed change of altering the duration is all that is necessary, and that Keith foresaw this in his design of this thread.  Next question to overcome... since the warning goes away as soon as there is user activity... do we want the user NOT to see this in the event they plug it in and forget to pay attention (instead of deal with their wallet coordinator while it boots) for the next 20s?  Truth is, they'll pay attention at least a few times and they'll already know that the sdcard can be removed.

easyuxd, 2024-07-18:
@jdlcdl 

Very thoughtful, as always. If the user HAD to perform a task here in order to proceed, I think it could warrant a persistent message. 

But as you said, occasionally missing the message is an edge case, and several seconds is long enough to catch it most times (especially considering the boot time is much faster now). At present, the UI is obfuscated with a non-critical message.

alvroble, 2024-07-19:
Thank you for your thorough comments. Maybe there should be added some personalization in the tools menu? I was thinking of some default behavior, let's say 5s timer. But if the user just wants just to let it show until he removes the SD card (maybe they just plug the device and forget for some minutes, but want to be sure they can remove it when getting in action) just having a checkbox in the settings for a persistent warning could cover this use case also. Maybe even this could be a blocking warning, so the user could force themselves to unplug the SD card each time.

Maybe we are making such a big deal of this, but I believe it’s worth considering. This is one of the user’s first interactions with the device each time they power it on.

easyuxd, 2024-07-19:
Love your recommendation, @alvroble. An advanced menu setting would be great. Everyone can be appeased that way. Options something like?

- 5 seconds (default)
- Persistent
- Until removed
- Off

kdmukai, 2024-07-29:
There was also talk at one point of having an SD card status icon, perhaps just on the Home screen, to indicate if the SD card was present or not (e.g. green SD card icon at top left, gray if it's removed?).

That might satisfy the most hard-core paranoids since they can just glance at that corner and reassure themselves via the icon that they haven't forgotten to remove the SD card. But it's quiet and innocuous so that less savvy users would barely even notice it.

It would help to think through the matrix of what types of different people we are designing around (first time noobs, experienced users, max paranoids) and what approach best suits each type w/minimal interference or confusion for the other types (i.e. there's almost definitely no one-size-fits-all perfect solution here, but probably a more-than-good-enough solution for everyone).

AND / OR there was also previous discussion about an option to **require** SD card removal before proceeding. Obviously the best choice for the max paranoids. Also probably makes the SD icon unnecessary. I can't recall if there was any reason why we didn't implement that as an option.

kdmukai, 2024-07-29:
In which case I'd modify @easyuxd's suggestion above to something more like:

* Reminder disabled (experienced user that isn't worried if the SD card remains (me!))
* 5-sec reminder (default)
* SD card removal required* (dedicated screen that doesn't let you proceed until removal; after that, obv no need for the reminder toast)

I don't think there's a need for an infinite persistent option for the toast.

---

*(probably need an escape hatch on that View if we know we're running Raspi OS (i.e. a dev environment) that doesn't actually support SD card removal)

easyuxd, 2024-07-29:
@kdmukai 
I'm on board. These 3 options cover all the bases nicely.

> It would help to think through the matrix of what types of different people we are designing around (first time noobs, experienced users, max paranoids) and what approach best suits each type [...]

Exactly. This highlights the needs for user personas. I'll start working on this.

alvroble, 2024-07-29:
I completely agree @kdmukai! 

@easyuxd let me know if I can help on this, as I have time to contribute more. I'll keep an eye on what you propose.

jdlcdl, 2024-08-27:
Today, someone in the main telegram group expressed interest in a settings choice that forces them to remove the sdcard when using seedsigner.

easyuxd, 2024-08-27:
I'm working on some user research (it will take a little while), but think it's safe to go with Keith's 3 options we've agreed on.

@alvroble Do you still want to tackle this PR? I can help with the fullscreen dialog for #3 if needed.

alvroble, 2024-08-27:
@easyuxd I will try to build something around. Regarding the fullscreen dialog, as soon as I have the logic flow ready I will submit a first UI iteration so you can then propose improvements.

alvroble, 2024-08-31:
@Semisol, the default behavior would be a 5-second timer, allowing anyone to change any persistent settings before this one. I don't really understand the problem you mention.

alvroble, 2024-09-01:
I'm thinking of two alternative solutions for this catch-22 situation:

1. Use the Raspberry Pi’s internal storage (e.g., /boot) to save a minimal config file that only stores the SD card setting. This requires modifying how settings are accessed and stored, and it might imply a slightly complex refactor.
2. Define a specific key combination or button press sequence during the full screen dialog to exit it. This would let the user access the UI and modify settings as needed. However it must be documented for the user, maybe in the dialog itself.

I believe the second one would be better since it doesn't mix the settings writing/reading, although maybe the first approach is a good alternative to write **all** the settings for the SeedSigner OS builds.

I'd love some input here from other devs so I can redirect the coding to one of these solutions (or any other one proposed)

easyuxd, 2024-09-01:
@alvroble Could it be as simple as what @Semisol proposed -- add a button to go to the settings screen for this toast? Maybe I'm not seeing the benefit of having a wait time or special key combo if the setting can eventually be changed anyway.

[image]

alvroble, 2024-09-01:
@easyuxd thanks for your comment. The problem here is that in my branch I actually use a fullscreen toast, so the remove SD card flow is unaltered. This would imply a different approach, as we'd talk about a view and not a toast. Maybe a more complex refactor. I'll check on it.

newtonick, 2024-09-05:
> @alvroble Could it be as simple as what @Semisol proposed -- make Option 3 (microSD removal required) a fullscreen dialog add a button to go to the specific settings screen? Maybe I'm not seeing the benefit of having a wait time or special key combo if the setting can eventually be changed anyway.
> 
> <img alt="remove-memory-card" width="240" src="https://private-user-images.githubusercontent.com/86452817/363503351-a21c6291-6de3-4137-9843-0636a42ddea0.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjU1NjYxNTEsIm5iZiI6MTcyNTU2NTg1MSwicGF0aCI6Ii84NjQ1MjgxNy8zNjM1MDMzNTEtYTIxYzYyOTEtNmRlMy00MTM3LTk4NDMtMDYzNmE0MmRkZWEwLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA5MDUlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwOTA1VDE5NTA1MVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTYwMDgzNGQzNjk0MzE2OThiNmJkMDUyOGYyZjgyNzBmMzQ1NTFjN2FiMzk2MDM1NzRkNjIyNDdjZDVhYjA3MjgmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.ZVrtDQBro1Ff_jMvImoh9tg0XkZRuiqShq4lzoobbBg">

I like this better for option 3 as well. Would be ideal if it jumped straight to the specific settings screen when the "settings" button is selection. Then when the user selects the back "<" arrow, the "Remove Memory Card" would re-evaluate the state of things and proceed accordingly. I'd have to dive into the code to see if jumping straight to a specific setting is possible.

[image]