SeedSigner/seedsigner #553: Sign Message tool not available under Tools

Purchase73, 2024-05-03:
It's possible to sign a message in Seed Signer (as long as you have that functionality enabled, under Settings, Advanced).

However, it appears the only way to do it is to just scan the QR code, from the main screen.

I expected the Sign Message functionality to be available right next to the Verify Address tool, under Tools on the main screen.

SeedSigner, 2024-05-03:
You can see a demo of this process here:

https://x.com/SeedSigner/status/1781073766300143725

Please note that if you're attempting to use this in conjunction with Ocean's payout verification process, there is a known issue with that process, described here:

https://github.com/SeedSigner/seedsigner/issues/550

Purchase73, 2024-05-04:
Thanks, that's a good demo of how to use it. I had a hard time finding it, though, I expected it to be right next to Verify Address, under Tools. I messed around with it for a while, then finally discovered I could just scan the QR code.

This is a UI only issue, obviously the functionality is there now.