SeedSigner/seedsigner #562: APK release

stevexyz, 2024-06-29:
Reading [positioning seedsigner for maximum impact](https://seedsigner.com/positioning-seedsigner-for-maximum-impact/), I was thinking that an apk release would be a good option, since almost everyone nowdays (even in the first world) have an unused old phone somewhere around, and even a very very old phone would have the power for seedsigner

alvroble, 2024-06-30:
I agree with you, an old and offline smartphone can make a good signing device. And this apk could even be made available as a hot-wallet alternative for newbies, doing two things at once while educating newcomers in private key management with low hardware costs. If managed well, this could be a good positioning strategy for SeedSigner.

MostafaAhmed189, 2024-07-01:
zs

MostafaAhmed189, 2024-07-01:
[image]

MostafaAhmed189, 2024-07-01:
dwed