SeedSigner/seedsigner #564: Custom Derivation address explorer not yet implemented

cezar1, 2024-07-04:
Hello and first of all thanks for this great product.

I'm hitting line 627:
[image]

What I'm trying to do is use my master seed phrase to derive nostr public address with custom derivation path `m/44'/1237'/0'/0/0`

Can I be guided a little bit on how it could be implemented? Then I'll propose a PR.

jdlcdl, 2024-07-05:
I don't know how far you are going with this, so I'll make some assumptions and try to help.

Assuming that you want to derive an identity that you can sign nostr messages with, you're following nip-06 to use a bip39 mnemonic and passphrase to get a bip32 extended private key and then deriving the nip-06 derivation path to get a new extended private key which would be capable of both signing messages for its public key and also to compute its public key via ec math.

Using existing seedsigner tools you're not going to get that far but you can get close by exporting the xpub of the same derivation path, or even of just m/44'/1237'/0' and then using another tool to derive the relative public key of m/0/0 (finishing the  full derivation) such as embit.  The public key will be encoded within the xpub.  You'd then convert the final public key to its x-only form as hex (for nip-01) or as an npub (for nip-19).  This would enable you to derive a public identity/address for nostr, but it won't be enough to sign messages.

I suspect that you're really looking for a way to manage nostr posts via a seedsigner modification.  In that case, it will be more work, but seedsigner already has most of the libs (if not all) necessary to do that.  It's less likely at this point in time that seedsigner is ready to manage nostr.  Even if it were capable to do so, not sure that there is sufficient demand, yet, for seedsigner to both store-of-wealth and freedom-of-speech tasks within the same code-base (the fear being increased attack surface whenever a codebase tries to do more than its primary goal).

@kdmukai did similar work, and even more, in an experimental branch back in early '23 (prior to Nostrica), so you may follow Keith and check out his branch here (especially his helpers/nostr.py module): https://github.com/SeedSigner/seedsigner/compare/dev...kdmukai:seedsigner:nostr_delegation  As well, check out his posts on the topic in nostr or X from the period of time spring-summer '23.  He's got this tool too which will be a great resource (github links and nostr id inside) https://nostrtool.com/

Do you know of the SeedSigner telegram group?  if so, there is also a SeedSigner-New-Devs telegram group that you might poke for ideas.

cezar1, 2024-07-07:
Thanks for the reply. Nostrtool is definetely what i was looking for. I'll check the telegram group.