SeedSigner/seedsigner #570: Signing messages that overflow raise "TextDoesNotFit" exception

jdlcdl, 2024-07-13:
Reported in the main telegram group, within the past hour:

[image]
```
Hello Guys! I mine via the Ocean Pool and wanted to set up the "Lightning payout". To do this you need to have a Bolt-12 address or sign this address. I use Sparrow Wallet with Seedsigner. When I want to sign the message from Ocean, this error message appears because the message is too long and the Seedsigner cannot display it. Just a suggestion to the community in case this could be fixed in the future. Best regards!
```

This happens when the text has nowhere to cleanly break to a new line.

I had resolved this in the past for an experimental multilanguage branch, and it was to simply enable overflow... but then the user would be signing a message that they have not fully viewed on seedsigner screen.

Lagrang3, 2024-07-15:
This is the same issue described in #550

jdlcdl, 2024-07-15:
> This is the same issue described in #550

Thank @lagrang3, I'll close this one.

closed 2024-07-15 by jdlcdl