SeedSigner/seedsigner #585: [Enhancement] Dialog for seed already loaded in memory

easyuxd, 2024-08-01:
**Problem:**

When the user tries to import a seed that has already been loaded into memory, the system automatically navigates them to the Seed Options screen. This can appear jarring and introduces potential confusion, as experienced when doing live demos.

**Solution:**

Introduce a new dialog that informs the user that the seed has already been imported: Display the fingerprint and options to either view the seed options screen (View seed) or go back to the entry point (Done).

**Proposed Design:**

[image]

alvroble, 2024-08-01:
Good point @easyuxd! From my perspective, I didn't thought this was a confusing behavior when I tried to load an already existing seed, but I see it can be a bit confusing for newbies (even more in a demo). However, I'm not entirely sure there should be added an additional step for the user with a new view. I was thinking that maybe a toast message notification when the Seed Options menu is loaded would be enough for the user, just to know that they already had loaded that seed. 

What are your thoughts on this alternative? I could begin setting up a PR to work on the solution we decide.

easyuxd, 2024-08-01:
Amazing proposal, @alvroble. This is a much better use case for toast/snackbar, since it would be non-interruptive. It would cover part of the screen, but that could be mitigated if it was dismissible on any input (maybe a 5 second timer too?).

An "informational" snackbar would be ideal if it exists. What do you think of the content? I was thinking it'd be helpful to provide the fingerprint if it fits...

[image]

alvroble, 2024-08-02:
That exactly what I meant @easyuxd! That informational snackbar looks nice. About showing the fingerprint in the toast, doesn't it look a bit redundant, since fingerprint is already on top?

I'll try to start building something around this issue over the weekend.

easyuxd, 2024-08-02:
@alvroble 
I was thinking it'd be helpful for verification purposes within the message, but I think you're right: It should be obvious enough considering the fingerprint is visible in the navbar and the user was just navigated here.

Regarding informational snackbar: I'm not sure if there's a current example of this component with a messaging icon. Maybe the microSD snackbar is the closest template. I have a messaging PR planned for the coming months that will refine the message types and provide usage guidelines, but I'm not sure when it will be ready (esp given my poor coding skills). Would love to get your input on that.

[image]

alvroble, 2024-08-04:
Love this, @easyuxd. I see that as more user options and alternatives are included, the usage of this kind of screen will be quite useful, so maybe it's worth considering a medium-size (?) refactor to include this as some sort of "messaging suite" and follow some usage guidelines of the different messaging options. I think is worth it and I see it as an insurance for the functionality growth that SS might have. Let's see what other contributors think.