SeedSigner/seedsigner #606: [Enhancement] Messaging Design Changes

easyuxd, 2024-09-09:
This is an effort to categorize and clean up all screen types. The changes are itemized below, and are informed by the [SeedSigner UX Guidelines](https://github.com/easyuxd/seedsigner-ux-guidelines) I published.

These design enhancements were previously socialized with @kdmukai, @SeedSigner, and @newtonick. The changes are synergistic with upcoming multi-language support, and so I wanted to outline the proposal in full, share my progress, and recruit programming help with the refactoring that is required but beyond my expertise.

I anticipate the refactoring of the menu and dialog screens will impact the current content and therefore may affect the translations.

## Overview

- Reorganized Dialog screens into clearer categories
- Added new dialog type: Dire Warning
- Visual clean-up for Dialog screens and Toast component
- Separated Menu screens into 3 layouts: Grid, List, and Bottom List

## Detailed Changes

### 1. Messaging components (Complete, [PR #605](https://github.com/SeedSigner/seedsigner/pull/605))

[image]

- New icons: info, sign
- New color: Dire Warning (#FF5700)

### 2. Messaging layouts and templatization (NEED HELP)

[image]

- Reorganize Dialog screens into clearer categories, add Dire Warning (separate from Warning and Error categories)

- Refactor dialog layouts (LargeIconStatusScreen)
	- Remove status_headline
	- Change status_color to "GUIConstants.BODY_FONT_COLOR"

- Templatize dialog categories:
	- Current classes: SuccessScreen, WarningScreen, DireWarningScreen
	- Proposed dialog classes: DefaultScreen, InfoScreen, SuccessScreen, WarningScreen, DireWarningScreen, ErrorScreen

- Toast categories to align with dialog categories
	- Refactor started by @alvroble: https://github.com/SeedSigner/seedsigner/pull/586

- Refactor menu categories: Grid, List, BottomList
	- [Grid](https://github.com/easyuxd/seedsigner-ux-guidelines?tab=readme-ov-file#menu---grid): no change
	- [List](https://github.com/easyuxd/seedsigner-ux-guidelines?tab=readme-ov-file#menu---list): 3 or more items with no descriptive content (add descriptive content area to layout)
	- [Bottom List](https://github.com/easyuxd/seedsigner-ux-guidelines?tab=readme-ov-file#menu---bottom-list): 3 or less items with descriptive content 

### 3. Messaging content

- Write more descriptive 2-word screen titles to replace generic titles
- Add descriptive content to Bottom List menus
- Requires oversight from @kdmukai, who wrote most of the original content

alvroble, 2024-09-10:
This is great, @easyuxd. Thank you for the work of grouping all the issues into one. It was needed so we can all row towards one direction now. I'm able to help with the dialog & toast categorization refactor, as I think those two should go together in the naming.

bitcoinprecept, 2024-09-10:
Fantastic work @easyuxd