SeedSigner/seedsigner #607: Camera Not Enabled Error with Pi5-IMX219

guilherme-salome, 2024-09-12:
Hi everyone, not sure if this is the right place to ask for help, if it isn't please let me know. 

I am a noob on raspberry pi and bitcoin in general, but trying to learn. I set up a raspberry pi zero following seedsigner's manifesto, but made a mistake when purchasing the camera. The one I have is a Pi5-IMX219 from Waveshare (https://www.waveshare.com/wiki/Pi5-IMX219). 

Problem: When trying to test the camera, I get a System Error with the message
```
PiCameraError
camera.py, 459, in _init_camera Camera is not enabled.
Try running `sudo raspi-config`...
```

Does anyone have experience with these problems and can point me in the right direction?

Thank you!

krizzi24, 2024-12-02:
I have the same problem. Is there a possibility to run a terminal on seedsigner-os?

jdlcdl, 2024-12-02:
Leaving a pointer to the seedsigner-os repo where I've added a similar issue.

When looking at that repo, can try something like: https://github.com/search?q=repo%3ASeedSigner%2Fseedsigner-os++imx219&type=code to get a hint of which config file changes might be necessary.

Issue in seedsigner os is: https://github.com/SeedSigner/seedsigner-os/issues/80

ghamy016, 2025-01-02:
dealing with same problem, any solution advised?

SeedSigner, 2025-01-02:
This error was thrown b/c the OP used a camera that is not currently supported by our software, the solution is to acquire a standard RPi camera, there are links to a few known good ones at the bottom of the page at https://seedsigner.com/hardware

ghamy016, 2025-01-03:
> dealing with same problem, any solution advised?

i had no daughterboard pi zero camera , just was not inserting it the right way, which is, insert into the pi board slot from botton not above, so problem solved!

pablopeu, 2025-02-16:
> Leaving a pointer to the seedsigner-os repo where I've added a similar issue.
> 
> When looking at that repo, can try something like: https://github.com/search?q=repo%3ASeedSigner%2Fseedsigner-os++imx219&type=code to get a hint of which config file changes might be necessary.
> 
> Issue in seedsigner os is: [SeedSigner/seedsigner-os#80](https://github.com/SeedSigner/seedsigner-os/issues/80)

made a modified build putting y (yes) instead of m for the settings mentioned but it did not work, more tweaks are necesary