SeedSigner/seedsigner #621: [Enhancement] Add option of address verification to XPUB export workflow

3rdIteration, 2024-11-03:
Basically just adding an additional screen to the workflow that allows the user to scan a receiving address after exporting an xpub. (To fully validate that they have imported it correctly)

The current workflow automatically returns to the home screen after showing the QR codes for export, the new workflow would just add an additional screen offering something like "Validate Receive address" and "Return Home". (The workflow for validate receive address would instruct the user to open their wallet software, select receive and then offer a button to scan the resulting address, which would just initiate the existing address validation workflow, pre-selecting the seed,etc, that was just exported)

You can see how Foundation passport implements this here: https://youtu.be/n9EPXCJWTB0?si=Akd_TlSTOkji86k5&t=539

I'm happy to knock up a PR for this in the coming weeks, but it's also a fairly simple issue if someone else wants to have a crack at it :)

fedebuyito, 2024-11-12:
Hi, @3rdIteration. This is a great enhancement, IMHO. I had to fix workflow you asking, but I am missing a bit detail in order to verification step does not ask for seed since it already is selected and on memory (I am cracking at it :-)).

3rdIteration, 2024-11-12:
Thanks :) So are you saying that you are coding it up and are stuck or just tweaking the suggested workflow?

fedebuyito, 2024-11-13:
Thanks to you! I have resolved suggested workflow but I have to work a bit hard because there were that passing seed_num across class calling, from XPUB export to verification screen since seed is already loaded and must not be asked. Flow tests have to update, too.