SeedSigner/seedsigner #626: Seed memory cleanse

dhruv, 2024-11-09:
I'm curious where in the codebase is an attempt made to cleanse the memory upon shutdown where the seed is stored (or otherwise)? Can you please point me to it? Happy to ask the question elsewhere if this channel is inappropriate.

SeedSigner, 2024-11-10:
RAM is a type of volatile memory, such that when power is removed, it resets to a default start state. So any information that is present in powered RAM is lost when power is removed from the memory. You can read more at: https://en.wikipedia.org/wiki/Volatile_memory .

SeedSigner OS and the SeedSigner app code run entirely in RAM on a Raspberry Pi; you can verify this by removing the microSD card once a signer has booted to the device's main user interface. The signer will continue to run normally b/c all of our code has been copied from the memory card into the Pi's RAM during the boot process. Later, when you remove power from the signer, data related to any keys you've created or used, or transactions you've made, is lost as the RAM resets to its default state.

If you have more questions, please visit our telegram community at https://t.me/joinchat/GHNuc_nhNQjLPWsS

closed 2024-11-10 by SeedSigner

kdmukai, 2024-11-10:
This research paper is the best source I've found for the speed of RAM data decay under different conditions:

https://citp.princeton.edu/our-work/memory/

Some notable takeaways:

At normal operating temps: _"We found that the dimensions of the decay curves varied considerably between machines, with the fastest exhibiting complete data loss in approximately 2.5 seconds and the slowest taking an average of 35 seconds."_

So for a typical SeedSigner user, the evil maid doesn't have much of a chance to get in there and attempt to read what's left of the data.

Next they tested recovery if the RAM is cooled to -50°C via a can of compressed air:
[image]
(A, B, C, and D are different test computers with different RAM)

In the most extreme case (machine B) while the RAM was in this cooled state, they were able to fully recover data as far out as 6 minutes and recover nearly all of it even 10 minutes later. They do an additional experiment where the RAM is then totally submerged in liquid nitrogen and were able to recover data an hour later. They theorize that _"data may be recoverable for hours or days with sufficient cooling."_

HOWEVER, note their procedure:
_"In each trial, we loaded a pseudorandom test pattern into memory, and, with the computer running, cooled the memory module to approximately −50◦C. We then powered off the machine and maintained this temperature until power was restored."_

So for a SeedSigner user, a RAM recovery attack would require you to LEAVE THE DEVICE RUNNING to give the evil maid the chance to cool it BEFORE powering off.

---

**tldr: RAM recovery attacks aren't a practical concern for SeedSigner users.**

The biggest risk is still: naked single sig (one key, no bip39 passphrase). Single sig + bip39 passphrase or multisig are much stronger approaches. I strongly prefer multisig. If you're not there already, set a goal to level up to one of those options as soon as you're ready.