SeedSigner/seedsigner #628: [Feature Request] Enclosure for WaveShare 19739 Battery HAT Extension

kokokale, 2024-11-10:
I'm not sure if there has been any previous discussion around adding on-board power to the SeedSigner or not, but I would personally really love the ability to go entirely wireless with this device.  Hardware and 3D modeling are not one of my skillsets, but I believe that with some minimal modifications we can extend existing SeedSigner enclosures to fit this WaveShare Power-Supply HAT: https://www.waveshare.com/ups-hat-c.htm (See images in product listing for an example of this HAT stacked with the 1.3" WaveShare display SeedSigner currently uses)

I can see a few things to consider here:  
- This HAT is sold with 4 mounting screws for the Pi.  Personally, I would love the rigidity that comes with this style mounting, but understand if the best approach is to have the enclosure handling this, as the enclosure I personally have does with the Pi Zero board and 1.3" display/controller HAT.
- I believe this HAT can be mounted and supply power through just the GPIO pins without any modifications to the existing SeedSigner hardware or software.  If this feature is to be adopted in future enclosure designs, I'd be happy to create an issue to add battery monitoring with a GUI display of battery charge levels, and assign to myself.
- I'm aware that this will likely involve moving the camera port in the enclosure, but I believe we have enough long enough ribbons on the currently tested cameras to just offset the port.  If keeping the port location fixed is a hard requirement, maybe we can find a slightly longer ribbon.
- I'm not aware of any excess heat this setup may generate, but it's possible we may need to include some ventilation slots in this new enclosure design.

Since this design is currently untested, I understand if we would like proof-of-concept before proceeding.  This request is mainly a request for comment and to gauge interest.  I am currently in the process of setting up my SeedSigner, and after everything is functional, I would be happy to purchase and test this hardware for myself and report back my results.

Rob0xFF, 2024-11-12:
Please see my enclosue here for reference, hope this might help. Battery monitoring would be great.

[](https://www.printables.com/model/1069037-seedsigner-enclosure-for-ups)

SeedSigner, 2024-11-12:
I appreciate your interest and creativity here, but the project will likely never officially support any kind of build that includes a battery. The assurances around a SeedSigner's stateless nature arise out of power being completely disconnected from the device, so that the data in memory can clear. Adding a battery also adds the potential for a trickle of power that could cause the device to retain information about private keys that have been created or used with the device.

kokokale, 2024-11-12:
Thank you for the response.  I understand the security concerns with this model, so I won't create any MRs to this project for battery support.

closed 2024-11-12 by kokokale