SeedSigner/seedsigner #630: [Discussion/Feature Request]

Rob0xFF, 2024-11-16:
Do you think it might be possible to store the seed entropy in a compact QR code but additionally scrambled with a PIN, let‘s say 4-8 letters/numerics. I know there is something like a 25th word already in the mnemotic phrase design, but it would add some kind of additional security for those with 12/24 word seeds who want to store them as QR in case someone could steal it. It would still be worthless without the right PIN.

On seed backup via QR, the user would define the PIN once, and he would be prompted to put in the PIN on each time the QR is read again.

Thanks for thoughts/consideration.

fedebuyito, 2024-11-18:
Hi. It would be like passphrase functionality?

Rob0xFF, 2024-11-19:
No, it would be a means to encrypt the seed QR to render it useless, if somebody finds it.

fedebuyito, 2024-11-19:
Ok, I understood. You are maybe talking about to a encrypted QR function has krux project, for instance? Very good idea.

kdmukai, 2024-11-23:
> PIN, let‘s say 4-8 letters/numerics.

PINs can be short on other hardware devices because they have built-in rate limiters (delay for how soon you can guess again) and eventually lock out options after too many wrong guesses.

A short PIN in this use case could just be trivially brute forced more or less instantly.

NACK

The only way to really secure it would be to use a strong passphrase (20+ upper, lower, digits, symbols). At which point you're better off using a bip39 passphrase.

It would also complicate the recovery process for legit users (e.g. your heirs).

If you are currently on a "naked" single sig (i.e. no bip39 passphrase), there is NO good long-term way to secure the seed/backups/etc regardless of which hardware wallet or approach you use (I'm not a fan of Shamir nor SeedXOR for this use case).

Naked single sig just isn't viable for long-term storage.

Rob0xFF, 2024-11-23:
Agreed about your thoughts on bip39 passphrase. It would not replace its use and usefulness. But: Imagine some evil burglar who has no clue you have a seed signer, store your seed as a QR code or that you even own bitcoin. He might find the QR, even scan it with his phone and nevertheless find nothing but nonsense in it. No hint it encodes your net worth.
On the other side, if the seed is not encoded, he might find the 12/24 words interesting, google for them and find out it has something to do with bitcoin.
These are just my thoughts, of course, I see your points about the passphrase.

alvroble, 2024-11-29:
If I understood you correctly, I think you've just given an actual use case for BIP-39 passphrase. The thing is that you can already create a BIP-39 passphrase that performs that function. Common recommendation is to keep seed words and passhphrase separate because of that "unlocking" function the passphrase gives.

kdmukai, 2024-12-12:
> On the other side, if the seed is not encoded, he might find the 12/24 words interesting, google for them and find out it has something to do with bitcoin.

This is not how either SeedQR format ("Compact" nor "Standard") work.

Scan these with your phone:

Compact
[image]

Standard
[image]