SeedSigner/seedsigner #633: Add Silent Payment Send Optionality as new feature

Gudnessuche, 2024-12-12:
Given the open-source nature of this project, and keeping with the cypher punk ethos, it's only prudent that Seed Signer had silent payment support as a hardware wallet.

BIP 352(Silent Payment) send option so far is only implemented by BITBOX-02 when it comes to HWWs. I believe that since this BIP has been merged into core, it only makes sense that Seed Signer lead in proliferating SP usage.

kdmukai, 2024-12-12:
I am eager to get SP support into SeedSigner. See my initial work here: https://x.com/KeithMukai/status/1790913869453418718

My understanding is that we'd need an [updated psbt spec](https://gist.github.com/andrewtoth/dc26f683010cd53aca8e477504c49260) to be completed before we could create a full integration on our end.

Josie will be presenting at my local Bitdevs meetup tonight so I'll make sure I get more details.

Gudnessuche, 2024-12-12:
[image]
Cool just seeing the above. I also want toknow how you intend on implementing the idea of Labels and Contacts as well as BIP 353 SP.

kdmukai, 2024-12-12:
My tentative answer is that it depends on what use cases emerge and what other tools support.

For example, "accounts" in derivation paths has largely fallen out of favor and most wallets don't even expose that functionality. So the default SeedSigner user flow also completely ignores accounts (however, since some account usage DOES exist, users can always specify a full custom derivation path).

If Labels end up being similarly fringe / unused, then so be it.

I'm not familiar with the Contacts concept.

For now the simplest, most basic SP implementation seems a good target for a first release.

Gudnessuche, 2024-12-12:
I agree. Pumped you and Josie are seeing tonight, should be positive.

LFG!!

fedebuyito, 2024-12-17:
Hi @kdmukai , I have proved your code and I could to verify that sp address Josie's branch generates  (https://github.com/bitcoin/bitcoin/pull/28453) match with yours, same as with silentium wallet that you verified (https://x.com/KeithMukai/status/1791134482130448480). Nice!

[image]
[image]

I am researching about BIP 352 since a few weeks and maybe I could help you or collaborate. Tell us more about your meeting with Josie, I think he is implementing this on Core here: https://github.com/bitcoin-core/secp256k1/pull/1519

Regards!

kdmukai, 2024-12-17:
It sounds like the updated psbt spec will take another couple months. Not surprising. So that's still a blocker for adding full support on the SeedSigner side.

The most interesting thing I learned was about the Labels. The spec describes them as an integer that you would increment for each new use (e.g. each tenant in your apartment building would get a different integer Label so you can keep track of who's paid you their monthly rent). But Josie said that you can use *anything* you want as a Label, so long as YOU know what scheme you used and are able to reproduce it. One example was if you're a big exchange, you could use each customer's unique customer_id.

The reason the spec only describes an incrementing int is that any wallet can trivially pre-compute the first, say, 100k Labels and include them in the scan when restoring a SP wallet. But if you use your own Label scheme, the onus is on you to set up your own system to scan for those Labels and find your SP utxos.

I asked if the bip could be clarified to include that additional info about Label flexibility and he agreed that it should be added.

andrewtoth, 2025-01-16:
Just FYI the psbt spec has been merged as [BIP375](https://github.com/bitcoin/bips/blob/master/bip-0375.mediawiki).

codemaster1104, 2025-04-19:
Hey Akshat this side, would love to work on this issue for SOB, should I create a new issue to give updates on my progress?
@kdmukai

kdmukai, 2025-04-19:
> Hey Akshat this side, would love to work on this issue for SOB, should I create a new issue to give updates on my progress? [@kdmukai](https://github.com/kdmukai)

It is unnecessary and inappropriate to "@" reference me in this context.

My goal is to support SoB students the best that I can. But you shouldn't be summoning me for attention unless there's a specific question that only I can answer. See all the info in the Devs' telegram group. The answer in there over and over again is: ask questions and get clarifications, but if you're looking for permission, you are totally misunderstanding how our project and most FOSS projects operate.

codemaster1104, 2025-04-20:
[image]

Hey, just tried to replicate [previous implementation](https://github.com/SeedSigner/seedsigner/compare/dev...kdmukai:seedsigner:bip352_silent_payments_basic_support).
Also did try to add support for psbt signing but am not sure how to test it as its not yet available in bitcoin core regtest method, any suggestions on how can I test it?