SeedSigner/seedsigner #648: Adopt new style guideline for capitalization: "Title Case" titles vs "Sentence case" buttons

kdmukai, 2024-12-29:
`TopNav` titles should be "Title Case" where every major word (and, for now, every minor word, too) should be capitalized.

[image]

Only aware of one English screen that needs to be fixed: https://github.com/SeedSigner/seedsigner-screenshots/blob/dev/en/seed_views/SeedAddPassphraseExitDialogView.png

---

But buttons should be "Sentence case": only the the first word is capitalized (though proper nouns like "SeedQR" should always retain their capitalization).

There are many buttons that need capitalization fixes.

---

Some proper noun judgment calls:
- "Xpub" always capitalized? I vote no.
- "BIP-39 Passphrase"? Probably yes because it's such an important topic to get right.
- "Native Segwit" / "Nested Segwit" / "Taproot"? Yes? They are crucial concepts, even if they're not always capitalized elsewhere in the bitcoin world.
- "Custom derivation"? No need to always capitalize.

---

related: https://github.com/SeedSigner/seedsigner-translations/issues/8

jdlcdl, 2024-12-30:
Yes, YES, 100%!

Guidelines that are strict and implemented in English will also be picked up by detail-oriented translators where applicable.  While I don't understand Spanish, I'm picking-up that this is getting discussed in that tg group too.  It's helpful as I'll make similar suggestions in French.

bitcoinprecept, 2025-03-14:
We have discussed that cleaning up capitalization will trigger impacted foreign language strings to become blank until updated. This topic is viewed as a nice to have, lower in priority than releasing next batch of languages.

Once the full self-service screenshot generator feedback loop process is will oiled and functioning, that will be an ideal time to tackle this style guide clean up across all languages.