SeedSigner/seedsigner #659: Feature request: QR Code encoder and decoder

Mikael-R, 2025-01-05:
It would be a great improvement to add both QR code encoding and decoding functionality to SeedSigner. This feature would allow users to both generate QR codes and decode them, enhancing the overall user experience. With the ability to encode QR codes, users could securely share information like addresses or messages, while decoding QR codes would allow users to read information embedded in QR codes, directly from the SeedSigner device.

## Proposed Solution:
* Implement a QR code encoder feature within the SeedSigner interface that allows users to generate QR codes by encoding text or addresses.
* Add a QR code decoder feature that enables users to scan and decode QR codes, displaying the extracted text or data on the SeedSigner screen.
* Both features should be easily accessible within the device, with an option to toggle them on or off based on user preference.

## Benefits:
* Increased functionality, allowing users to both generate and decode QR codes securely on the device.
* More versatile and user-friendly for scenarios requiring secure QR code exchange or information retrieval.
* Streamlined user experience with everything handled on the SeedSigner device without needing external tools.

trinks-slam818, 2025-01-05:
Hello, we tried to solve the issue.

This is what we did:

Implement QR code encoding and decoding features in SeedSigner to allow users to generate and scan QR codes directly on the device. This enhances the functionality and user experience of SeedSigner.

You can review changes in this commit: https://github.com/trinks-slam818/SeedSigner-seedsigner-659/commit/57b0a2021b64d5a9eaf40551677f15c21d9e0055.

> [!CAUTION]
> Disclaimer: The concept of solution was created by AI and you should never copy paste this code before you check the correctness of generated code. Solution might not be complete, you should use this code as an inspiration only.

---

Latta AI seeks to solve problems in open source projects as part of its mission to support developers around the world. Learn more about our mission at https://latta.ai/ourmission . If you no longer want Latta AI to attempt solving issues on your repository, you can block this account.

seedcard, 2025-01-07:
I would support this feature. A generic QR code generator/decoder would be a a useful standalone tool and increase the general functionality of the SeedSigner device. I would suggest it could be added under Tools. 

The QR code could be engraved and stored on a SteelQR plate (see https://vulcan21.com/steelqr/) by including the same 'transcribe' functionality as the existing SeedQR output.

jdlcdl, 2025-01-07:
Feels like a basic air-gapped building block.

I currently use @pythcoiner's SeedQReader for this functionality.  Having read that code, I know that it remains on my computer (which is online and therefore valuable) instead of being shipped off to who-knows-where in the cloud, to be sold/stored/tracked.  But this is useful for inspecting and even preparing qrcodes in a pinch... with at least arguable confidence that privacy is preserved.

BUT... being able to do the same on a stateless air-gapped freedom-ware seedsigner would be preferred in some cases.

I like!

lorenzyannick, 2025-04-13:
I guess it's also usefull to store passphrases as QRCode. I hope so

kdmukai, 2025-05-31:
@0nkel-Bob the best practice for single sig is to use a bip39 passphrase. see: https://github.com/SeedSigner/seedsigner/issues/290#issuecomment-1369075791

Basic "naked" single sig (no bip39 passphrase; i.e. what everyone starts with) isn't viable long-term if you need it to protect you from every possible attack vector or failure scenario, no matter how you store it or access it.

SeedQR + passphrase is the same effective security as a bip39 mnemonic in any form (words, SeedQR) + bip39 passphrase. And whenever possible, the most standard, most common approach should be the preferred direction. Thus our strong preference for bip39 passphrase.

So while I agree with your statement, rather than trying to tack on a so-so solution to naked single sig, we strongly encourage:
* Creating a new single sig + bip39 passphrase setup
* Migrating to multisig (collaborative custody or full self-custody)
* Eventually perhaps miniscript, FROST, etc.

kdmukai, 2025-05-31:
> I guess it's also usefull to store passphrases as QRCode. I hope so

@lorenzyannick see: https://github.com/SeedSigner/seedsigner/issues/71#issuecomment-913782914 and the rest of the discussion in that thread.

Bicaru20, 2025-06-05:
Hi, what would this feature be used for? SeedSigner already lets you encode and decode QR codes for addresses, seeds, etc. Sorry, but I’m not sure I understand why we’d need an additional encoder or decoder for text or other types of data.