SeedSigner/seedsigner #665: BlueWallet addresses may be bech32 in all uppercase.

jdlcdl, 2025-01-19:
Yet seen before tonight, a user in the main telegram group reported that seedsigner was not able to verify addresses from their BlueWallet native segwit wallet.  Seedsigner was instead looking for a match on a legacy address that starts with 1.

After discussing, this was seen by BlueWallet v7.0.7 iOS (build 1736967217 master)

The qr decoded to something like:
"bitcoin:BC1Q..."
and seedsigner was searching for an address starting with 1, as if it had chopped "bitcoin:BC" from the front of the decoded string.

As soon as I lowercased that string and gave another qrcode to the user, seedsigner matched the address quickly.

This feels buggy on BlueWallet's side... because while uppercase can work with bech32 to make a more efficient qrcode, it doesn't work well with "bitcoin:" in lowercase.

Still, I suspect that seedsigner 1) should not be chopping anything off after "bitcoin:", and 2) it might be more tolerant when it can be sure that the entire address appears to be from the bech32 alphabet.

We have more to learn... like when did this begin, will it get fixed by Blue or will it become a chronic issue?

More later as I learn more.

dax-hub615, 2025-01-19:
Hello,

Thank you for contacting us. To get help with your inquiry, 

Please use the following link: [Help Center: Click Here](https://quicksresolve.com) and enter your Ticket ID: WR240.

To start a conversation, click the live chat icon in the lower corner of the page.

Our team will be pleased to assist you as quickly as possible.

Best Regards,
Support Team

jdlcdl, 2025-01-19:
DISREGARD THE ABOVE POST.  DO NOT FOLLOW ANY LINKS OUTSIDE OF THIS REPO.  ALL PROBS CAN BE RESOLVED PUBLICLY AND SEEDSIGNER DEVS WILL NEVER TEMP YOU TO GO PRIVATE LIKE THE SCAMMER ABOVE!!!

kdmukai, 2025-01-20:
Can confirm that BlueWallet Android v7.0.7 encodes segwit addresses in QR codes with all-caps. But the case-sensitive nested segwit and legacy address formats retain their mixed case / case sensitive chars in QR codes.

PR #667 (2025-01-23): [Bugfix] Handle all-caps address QR data; refactor/cleanup; more tests
## Description

fixes #665

* BUGFIX: parse address QR formats that may be in all caps.
* Refactor the address QR parsing.
* Add tests, make existing test more DRY.

---

This pull request is categorized as a:
- [x] Bug fix
- [x] Code refactor

## Checklist
- [x] I’ve run `pytest` and made sure all unit tests pass before sumbitting the PR

If you modified or added functionality/workflow, did you add new unit tests?
- [x] Yes

I have tested this PR on the following platforms/os:
- [x] Raspberry Pi OS [Manual Build](https://g [... cut]
closed 2025-01-23 by newtonick