SeedSigner/seedsigner #674: Feature Request: View the imported multisig output descriptor

TrentOmega, 2025-01-24:
### Request
Currently when importing a multisig output descriptor (via the Scan function or Address Explorer) it only displays the policy and finger prints of the signing keys. The request is for the "Descriptor Loaded" screen to also display the xpubs (BIP32 versions) of each signing key, along with the corresponding Y/Zpubs (SLIP132 versions). Showing the derivation paths is a nice to have.

### Motivation
This would allow us to verify that the actual co-signer xpubs are what is generating the addresses. The signing key's finger prints don't do this, as they are essentially detached from the xpubs, and could be anything. Without this function, a compromised coordinator could give us xpubs we don't control, and there is no way to verify them against what our Seedsigner, or other signing devices, are telling us what they are.

### Implementation
On the descriptor Loaded page, each signing key's finger print could be followed by the xpub and Y/Zpub, using a joystick scroll down to view them all. Maybe an accept right joystick, or an "Accept" button at the bottom. The xpub should come straight from the descriptor and the Y/Zpub would just need to be converted from the xpub. The derivation path, if implemented would come straight from the imported descriptor.

An alternative implementation would be to show a partial view of the xpub's and Y/Zpub's. This may make the UI a little cleaner, but less informative.

SeedSigner, 2025-01-27:
Can you confirm I understand the concern correctly -- the attack would incorporate a malicious descriptor, presumably contrived at the time of wallet creation, that incorporates master public keys that don't match the master private keys provided by the user during initial setup?

TrentOmega, 2025-01-28:
Correct.

TrentOmega, 2025-01-28:
My current work around for this is to receive bitcoin to an address on the wallet, then attempt to sign a transaction that has that as an input. This would be done with all Seedsigner devices in the multisig wallet. If the output descriptor, provided by the coordinator, doesn't contain a master public key that was derived from the master private key on the Seedsigner, then it will fail on signing.

fedebuyito, 2025-02-27:
Hi, @TrentOmega. I'm trying to find a workflow approach that fits with what you asked for. I want to know if this could help:

From the multisign descriptor load view, an option button for key details was added. From there, a list of active buttons appears (one per each key in the descriptor) with animated scroll showing all the complete key info (contained in the descriptor). This would enable different models of multisign with n/m schemes. 3/5 descriptor example:

[image]

Selecting one key will allow the user to step to a view with SLIP32 data (similar to the xpub export view, but adapted to this requirement).

[image]

What do you think?

TrentOmega, 2026-03-22:
This is exactly what would fix the issue.

oren-z0, 2026-08-15:
If the seed for the Xpub was loaded in the session, please show a label "YOURS".
Otherwise, show a label "COSIGNER".

Trezor shows this information when checking a Receive Address on the device.
It would make it easy to verify all the xpubs on all hw wallets, so the possibly-compromised coordinator app doesn't try to trick us into sending Bitcoin to a multisig wallet where the hacker holds one of the keys.