SeedSigner/seedsigner #679: [Discussion/FeatureRequest] Add new languages for BIP39 wordlist

polespinasa, 2025-02-11:
With the release of the translated versions, BIP39 word lists for those languages could also be added.

In my opinion the non-standardization of the language for BIP39 is a mistake in general and can bring problems to less experienced users. For that reason my proposal is to be able to activate in “advanced options” the use of the language chosen for the firmware also as language for the BIP39.

dulcedu, 2025-02-14:
**Reducing Errors in Seed Phrase Backup for Non-Native English Speakers**
For non-native English speakers, using an English-only BIP39 word list increases the likelihood of transcription errors when writing down seed phrases. If users do not understand English, we tend to  miswrite words, confuse similar-sounding terms, or make spelling mistakes. 

Example:
A Spanish speaker backing up their seed phrase in English might miswrite "piece" as "peace" or "despair" as "desperar" (a false cognate in Spanish). Since BIP39 requires exact word matching, a single mistake makes the wallet unrecoverable unless they later manually check the English word list—a process not all users are aware of.

0nkel-Bob, 2025-05-30:
Would it be possible to simply use the BIP 0039 numbers as an option to enter the words?

kdmukai, 2025-06-01:
@0nkel-Bob see: https://github.com/SeedSigner/seedsigner/issues/211

tldr: YES, that is an option. However, in this context, it would still be vitally important to know which wordlist was intended (English vs Spanish etc).

Any backups should be very careful to document both the number/index and the word itself (which still doesn't help, say, Arabic users who might not be familiar with English letters and words). It would be a scary error-prone risk if users only wrote numbers down ("1101" accidentally written as "1011" or a poorly written "5" that looks more like a "6" or a "1" that looks like a "7").