SeedSigner/seedsigner #681: [dev tool Proposal] Automatic and Custom Environments for Emulator Integration and Testing Suite with VENV

fedebuyito, 2025-02-11:
### **./tools/dev-tool.sh** fedebuyito/seedsigner@dev-tool https://github.com/fedebuyito/seedsigner/tree/dev-tool

### **Overview**:

I was coding a single-command tool in Bash to automate the building of virtual environments, with packages and sub-scripts, to integrate @enteropositivo's seedsigner Emulator and run Testing Suite for development tasks in separate Python vens.

### **Motivation:** 
Based on past discussions in the SeedSigner Telegram group (https://t.me/seedsigner_new_devs/9280) and the following PRs: https://github.com/SeedSigner/seedsigner/pull/668 and https://github.com/SeedSigner/seedsigner/pull/556, I have gained valuable insights that have shaped my approach to coding, testing, and reviewing. These experiences have helped me understand better practices and improve my contributions to the project.

### **The Goals were**:

- Using standard commands to make it easy to use a single .sh script to install customs venvs for each case, initially working on Debian/Ubuntu Linux and RaspberryPi OS.
- Isolating each development job including writing/probing code, running tests, switch branches, packages, and Python versions in dedicated venvs to avoid to dirtying any Seedsigner clonned repo by constantly activating/desactivating these venvs. 
- To be a pre-draft for a future PR if be useful for better devs productivity on Seedsigner

### **Instructions**:

You can test this script I've generically named `/tools/dev-tool.sh` included in my own fork. Clone the repository on dev-tool branch (fedebuyito/seedsigner@dev-tool https://github.com/fedebuyito/seedsigner/tree/dev-tool) and run the script as follows (from any folder, but ensure you are in the Seedsigner repo directory you are working on):

_**Usage: `./tools/dev-tool.sh` (`dev`/`test`) (`venv_name`) (`python3.x`)**_ 

First argument (venv mode): dev (default) for working with the emulator and code, or test for running test suite.
Second argument (venv name): custom name will be installed in /src or /tests depending on whether dev or test mode is chosen.
Third argument (venv Python ver.): to specify version you want to work with/test (e.g. `python3.10` `python 3.12`), in case you have more than one installed on your system.

**_Example 1 (dev mode): All three arguments can be omitted with `-` mark as follows:_**
 
~/seedsigner $ `./tools/dev-tool.sh - - -`

or simply,

~/seedsigner $ `./tools/dev-tool.sh` 

This will install a "dev mode" venv called DEV_VENV (with active Python version on your system), with the emulator and packages included. Then you only must to execute:

~/seedsigner/src $ `source DEV_VENV/bin/activate` (to activate venv)
~/seedsigner/src $ `python main.py` (to run seedsigner)

[image]

Done! (Don't forget to issue `deactivate` to exit the virtual environment when you finish your work)

**_Example 2 (test mode): You could to use the script for run test suite and to set a venv name._**

~/seedsigner $ `./tools/dev-tool.sh test 311_VENV`

This will install a "test mode" venv called 311_VENV, with packages and `pip install -e .` command included. So you only must to execute:

~/seedsigner $ `source ./tests/311_VENV/bin/activate` (to activate venv)
~/seedsigner $ `./tests/run_full_coverage.sh` (to run the test suite)

[image]

Done! (Don't forget to issue `deactivate` to exit the virtual environment when you finish your work)

**_Example 3 (dev mode for custom Python version): If you have any alternatives Python version installed, you could execute:_**

~/seedsigner $ `./tools/dev-tool.sh dev - python3.12` 

And this will install a "dev mode" venv (called DEV_VENV, because name omitted with `-`) on 3.12 Python version, with the emulator and packages included, too. After that you only must to execute same two commands of example 1 (`source DEV_VENV/bin/activate` and `python main.py`) to run seedsigner on this environment (and `deactivate` when you are done)
.

### **NOTES:**

**_Note 1. Activating / Desactivating Virtual Environment:_**

Each time you want to exit from venv you are working on, simply execute: `deactivate`. On "dev mode" venv, you could to verify status of your Seesigner local repo was not altered (at least you has made some custom dev change) and only there are untracking files (from emulator):

[image]

 You can check the repo status just after you activated venv, too:

[image]

[image]

And you will see how source files have seen "backuped" (see untracked and hidden `.file.py`) and provisional links (see `typechange: file.py`) to original ones have seen creating (being a safe emulator integration).

_**Note 2. Some results from my own tests and use of this tool:**_

- Seems to works well on Kali GNU/Linux Rolling (live) 2024 (x86_64) and Raspi-OS Debian GNU/Linux 12 (bookworm) (Pi4B).
- Emulator only works with Seedsigner 0.7.0 release on python3.10.
- From version 0.8.5 (the latest release as of today), the emulator works with a minor fix included in this PR: https://github.com/enteropositivo/seedsigner-emulator/pull/13. Until this is merged (or bug fixed) script will clone that version from my repo.
- Script aims to be simple and portable (as far as I have been able to test, only needs to use basic dependencies: e.g. git, rsync, python, venv).

_**Note 3. Missing:**_
- Additional argument for custom package selection.
- Refactor and translate everything into English, improve debugging.
- Raspi-OS full integration for optional hardware selection (ST7789 module and/or picam).

_**Note 4. Possible future enhancements (the tool could become more complex):**_

- Save changes (if there are any) on venv deactivation (for future recovery upon activation) and reset to original files from the repository.
- Use external tools (e.g. pipenv, nix or poetry) to package reproducibility.

_**Note 5. This is a prototype or concept probe designed to make the developers' jobs easier and safer for this project. It would be helpful if anyone could test it on other platforms or Linux/Mac versions. Any suggestions, bugs or comments will be welcome.**_

Regards!

polespinasa, 2025-02-12:
I tested this script to run the virtual seedsigner simulator and works perfectly fine.
Would be good if the script activates automatically the venv and launches the simulator.

Environment: Ubuntu 24.04.1 LTS

Bicaru20, 2025-03-28:
I tested the script using the example 3. When executing the command `./tools/dev-tool.sh dev - python3.12` I had no trouble. However, when I executed the `main.py`, I had a couple of problems (before executing `python main.py` I activated the environment):

1.  First, I had a dependency problem: `ModuleNotFoundError: No module named 'tkinter'`. I was able to easily correct that by running: `sudo apt-get install python3-tk`.
2. Then I was able to execute the `main.py`, and it looked like it was working fine, I even could interact with the emulator. But in the terminal it appeared another dependency issue: `ImportError: Unable to find zbar shared library`. Again, with just a command, I was able to fix it: `sudo apt-get install libzbar0`

I am posting this in case anyone has the same problem and to know if I did something wrong.

Environment: Ubuntu 24.04.2 LTS

`

fedebuyito, 2025-03-31:
> I tested this script to run the virtual seedsigner simulator and works perfectly fine. Would be good if the script activates automatically the venv and launches the simulator.
> 
> Environment: Ubuntu 24.04.1 LTS

Thank you, Pol, for your feedback. 

Yes, this script could automatically activate the venv, but since it is designed to work with multiple venvs (e.g., for different versions of Python, suite tests, or dev mode), it might be more practical and safer for the user to manually activate/deactivate each venv according to their needs. Activating the venv involves creating some linked files, which must be properly released using the deactivate script.

Regards. -

fedebuyito, 2025-03-31:
> I tested the script using the example 3. When executing the command `./tools/dev-tool.sh dev - python3.12` I had no trouble. However, when I executed the `main.py`, I had a couple of problems [...]

Thank you @Bicaru20 for your feedback. Some comments about them:

- I experienced problems like that (ModuleNotFoundError: No module named 'tkinter') when I wanted to perform an altinstall of another version of Python that comes with my distro (Kali Linux Live). Before running the configure-make-(alt)install commands (see https://docs.python.org/3.10/using/unix.html#building-python), some dependencies must be present on your system to install Python with the required options (like _tkinter or libssl_). These dependencies can also be installed:

`sudo apt-get install build-essential pkg-config libssl-dev tk-dev` (last could works instead of python3-tk)

- Regarding the ImportError: Unable to find zbar shared library, in this config file: https://github.com/SeedSigner/seedsigner/blob/dev/.github/workflows/tests.yml, for the CI tests, it can be noted that it runs on Ubuntu and that sudo apt-get install libzbar0 is issued, so what you did makes sense.

- I’d like to add this dependency as well, which might be necessary to fix an issue when QR codes are displayed:

`sudo apt-get install qrencode`

- Nice! All this can be useful for tracking the list of modules and dependencies required to run the emulator and suite tests for different operating systems.

Regards. -