SeedSigner/seedsigner #685: `SeedTranscribeSeedQRConfirmInvalidQRView` cuts off pixels from the bottom of the view for Dutch translation

rachid-o, 2025-02-24:
We just translated the text to Dutch `Your transcribed SeedQR could not be read!`. The screenshot generated is attached. If you look closely you see the bottom rows of pixels are missing. It's minor, but because the `Q` is a bit of a special character, it is now rendered as `O`.

I think it's a small rendering bug because there's sufficient space left on the screen. Not sure if this is only in the screenshot generator or also an issue on a real device.

[image]

kdmukai, 2025-02-24:
My guess is that the trimming logic in `TextArea` after it supersamples the text is only considering the text's "baseline" rather than the true bottom (can't recall what PIL calls that). Though if that was the case, I'd expect other multiline text to similar crop any below-baseline pixels (e.g. "p", "q", etc).

Might also be a difference between when the `height` is explicitly provided vs when it's auto-calculated.

Either way, this is something I'll need to look into (others are certainly welcome to it, but the `TextArea` stuff in general is pretty hard to work through so the path of minimum frustration is just to have me figure it out).

kdmukai, 2025-05-07:
Bug has been identified (my fault!). Working on a fix.

[image]

PR #751 (2025-05-25): [Bugfix] `TextArea`: Proper detection of chars rendering below baseline on final line of text
Fixes #685 

## The Problem

The bottom line of a multiline `TextArea` would only properly handle specific hard-coded characters that render below the baseline. The char list missed capital "Q". Worse, Asian fonts seem to render all characters a bit below baseline and none of them were properly handled.

[image] [image]

[image] [image]

[image] [image]

[image] [image]

---

## The Fix
The `TextArea` is already doing a number of `font.getbbox()` calls to figure out where to place line breaks in
closed 2025-05-25 by newtonick