SeedSigner/seedsigner #689: Add Codex32 Support (Please)

kiwihodl, 2025-03-06:
[image]

**You're a wizard, Seed Signer**.

Codex32 is a method of generating a bitcoin key(s), where you can split it using SSS (Shamir Secret Sharing). There is currently one wallet that has support for these keys: 

https://github.com/BenWestgate/Bails

I believe, due to SeedSigners DIY fashion, where it's users want to verify as much as they can, this style of key generations is idealistically compatible. Why is it important further? I'll leave it to Blockstream's team, it's [Codex32] creators, to explain the trade-offs further:

"How does Codex32 keys compare to BIP39?

BIP39 is a train-wreck of a protocol with the following issues, shared with neither codex32 nor SLIP39:
- BIP39 is encoded using 11-bit words, making it extremely hard to convert it to any other format without use of computers.
- BIP39 has an 4- or 8-bit checksum, which is too small to provide meaningful protection against random errors, and smaller than a single word, meaning that the "checksum word" also contains key data and so cannot simply be dropped.
- Compounding this, because you can't drop the last word, BIP39 encodes 128-bit secrets in 132 bits and 256-bit secrets in 264 bits, so the data doesn't fit in normal data containers (e.g. codex32-encoded secrets).
- BIP39's checksum uses multiple SHA2 iterations so it cannot be verified without the use of electronic computers or multiple days(!!) of hand computation.
- BIP39's checksum provides zero protection against adversarial errors. It is easy to change even a single word of a BIP39 phrase in a way that it will appear to be valid and simply have no coins on it.
- Consequently BIP39 cannot provide any error correction capability.
- BIP39 seed words are converted to BIP32 seeds (what your wallet actually uses) using SHA-512 to extend the words to 512 bits, which is completely unnecessary and prevents you from simply converting the words to a BIP32 seed and forgetting about BIP39 entirely.
- BIP39 does this by hashing the words themselves even though it supports wordlists in multiple languages and provides no in-band way to indicate which language is being used. Every part of this is an independent mistake in the protocol design.
- I do not mean to disparage the authors of BIP39, who are intelligent, well-meaning, and have all made great contributions to the space. But BIP39 was designed in 2013 when the Bitcoin ecosystem as a whole was lacking the institutional knowledge that we take for granted today, and you can tell.

Unfortunately BIP39 is by far the most widely-supported mechanism for producing BIP32 seeds today, both because of its advanced age and because the aforementioned protections against converting BIP39 words into anything better." - https://secretcodex32.com/faq/index.html

Repo:
https://github.com/BlockstreamResearch/codex32

You can see pertinent resources here for wallet developers:
https://github.com/BlockstreamResearch/codex32/blob/master/docs/wallets.md
https://github.com/BlockstreamResearch/codex32/issues/57

SeedSigner, 2025-03-06:
Thank you for opening this for consideration, I will be looking at some of the material above.

alvroble, 2025-03-06:
Hi guys! I have a draft PR (#636) which can serve as a proof of concept of SSS for SeedSigner. Maybe I could set up something up to try and build also Codex32 import and/or export.

kdmukai, 2025-03-07:
@kiwihodl can you tldr what the inputs would be? I've only briefly flipped through the physical workbook and that was quite a while ago (TABConf 2023...?). And speaking of which, how long would a Codex32 + SeedSigner workflow take for a user? The workbook looked like it was a couple hours of work but I don't know how much of that would be offloaded to the SeedSigner's computation in this case.

So:
* What user inputs to the SeedSigner?
* Roughly how long would it take the user to prep those inputs?
* Outputs are Shamir shares? If so, same wordlist as Trezor's?

Mild apologies that I'm not going out and just reading the docs myself, but while I'm mostly open to the idea of adding this, I'm not yet interested enough to invest the time to learn all the details. Just too many different rabbit holes in bitcoin so I try to limit how many I dive into.

stachrom, 2026-08-19:
any update on codex32 integration for Seedsigner?

float00, 2026-08-29:
> For import: its threshold amount of codex32 strings (48 characters). The "ms1" prefix can be assumed making it 45 bech32 characters. For T-1 subsequent shares the header (threshold & identifier) can be prefilled as all shares must match.

Above obviously relates to inputing the text codes, but one more upside to just assuming the "ms1" prefix: it shrinks QR one version, (assuming you want to QR encode it).

QR alphanumric mode fits 47 characters at version 2 with error correction L (272 data bits). A full 48 character share needs 277 bits. Strip "ms1" and you're at 45 characters / 261 bits, which fits.

  | payload | chars | QR Verison | Size |
  |---|---|---|---|
  | full ms1… | 48 | v3-L | 29x29 
  | ms1 stripped | 45 | v2-L | 25x25 |

BIP-93 even makes reference to uppercased bech32 being suitable for the QR alphanumeric charset.

Question for @benjamin-jarvie: was this issue meant to cover import only, or generation too?

If generation, then nice thing for SeedSigner I assume much of the transcribing to QR is reusable. 25x25 is the 12 word Standard SeedQR size, so the zoomed transcribe UI already handled.

BenWestgate, 2026-08-30:
@float00 I don't recommend dropping the prefix. If you want to fit the data in smaller QRs read my discussion on blockstreamresearch/codex32 I was able to fit codex32 shares in 21x21 v1 QR codes but its lossy or restricts to identifiers that start with QR and only threshold 2 and 3

float00, 2026-09-05:
Nice work on getting to  21×21.Deriving the index to save 5 bits is clever. My aim was much humbler. Just want 25×25 instead of 29×29. Seems unfortunate bip-93 targeted 48 chars for compatabiltity with common 12 word steel plates. 277 bits is just 5 bits over v2-L's 272.

BenWestgate, 2026-09-06:
Deriving the index from a 130-bit may be safe IF the indicies are chosen randomly like I do in `codex32 create` from the https://github.com/BenWestgate/python-codex32/tree/reviewability-v1
Branch of my codex32 implementation. I'm getting it polished up as we're adding it to BIP93 as a complete reference.

Some of the ideas in that thread like deriving the padding from 16 bytes is NOT safe and I realized can leak a few bits of the secret.

Probably for v1 21x21 compact CodexQR we may derive the indices from the payload or may not for more backwards compatibility with existing shares, then support threshold 0, 2 and 3 and with the bits left encode some identifier, hopefully 2 characters.

Its on my schedule to spec it out like Compact SeedQR but after the CLI and GUI for the reference is done.  For now, I'd drop a character of checksum to fit 48 alphanum chars in a v2 25x25 QR, this is easily corrected by ECWs marking the last character of scanned QRs as an erasure.