SeedSigner/seedsigner #692: Feature request to show a tick next to each fingerprint that has already signed a multisig PSBT.   123abc12✅

Jaffasoft, 2025-03-18:
Request to be able to turn off the SeedSigner icon floating all over screen. It is good and sometimes does get distracting for me a few times in the dark ect or during day trying to work. Setting to make it static in center of screen and or a black screen too. Maybe so ppl also can't know that it is a "Seed Signer" and look up investigate what i'm doing with it using bitcoin etc etc seeds rabbit hole bla bla etc. Animosity.  

Second request. As discussed in SeedSigner teligram recently. 

When loading a few seeds and fingerprints for a Multisig and trying to remember which ones you have already used to sign would be informative to see a tiny tick next to any seed that has already signed this existing transaction. It is not a big deal just i got a little confused as a first time user and it would of indicated to me which ones i had already used as i couldn't remember in the chaos. And choosing ones that had already signed was wasting time. 

846bdf1b
615ab1c2✅
35babc54✅

Taking this concept one step further: If they were ticked as such below (three in this case) perhaps also in a different screen titled 'Create Wallet Address' and an option menu to Show m of n Multisig addresses. Obviously if user only selected one seed second option would to be select show Single Sig Address and it should be able to click down through to address 10-20 or enter 200 etc or skip 10 something similar. But mainly the objective here would be to be able to enter a seed or seeds in SS and get a QR or an address to pay into. This is big and a good step that user then does not have to put any xPubs in any online wallet such as Sparrow Wallet etc. Though then user is trusting 100% that SS has generated the correct address and i guess not verifying the address with an open sparrow wallet or other online wallet. Ppl might have offline PC ect with SW but a lot do not and or use phones etc. Obviously no way of SS to know what previous addresses have been used or spent or have UTXOs etc but user can discern this info themselves and or be prompted with a warning to check for unused address etc. 

846bdf1b✅
615ab1c2✅
35babc54✅
'Create Wallet Address'

Even if could just put one seed in SS and be able to generate a QR of an address to pay into. This could be taken direct to the exchange screen to scan the QR code in or into a PC and copy paste into the place sending Bitcoin from. Non xPubs are ever online or temporarily online in a wallet or such and no privacy concerns.  Doing this it might as well have the function to put a single sig or Multisig wallet descriptor up on screen QR code so it can go into a wallet sometime as well. 

I realise there is a fundamental possible flaw to this idea as the descriptor and xpubs need to be stored and saved by the user initially setting our any wallet so it is a "good known" descriptor to check validate addresses feature already present in SS and a descriptor QR for my suggestion is too large for user to draw a QR code. So it a bit of a catch and may not work well. 

Thanks,
Regards,
Jaffasoft

fedebuyito, 2025-03-19:
Hi @Jaffasoft 

1) An option to disable the screensaver could be added to the advanced settings

2) Marking a fingerprint as used, I think, would be useful to do with a txid, as you might want to sign several PSBTs.

[image]
(sorry for my poor art)

I think a way to unmark would also be desirable. Additionally, this mark could be necessary for single-sign PSBTs—why not?

3) 'Create Wallet Address': Do you mean generating a multisig address without a wallet descriptor? If so, you would need to select the script and derivation path as additional steps.
> Though then user is trusting 100% that SS has generated the correct address and i guess not verifying the address with an open sparrow wallet or other online wallet. Ppl might have offline PC ect [...]

This must be the user's responsibility.
 
4) 
> Doing this it might as well have the function to put a single sig or Multisig wallet descriptor up on screen QR code so it can go into a wallet sometime as well.
> 
Do you mean something like an xPub export option but for multisig wallets? In other words, exporting the wallet descriptor created in the feature mentioned in point 3?

> I realise there is a fundamental possible flaw to this idea as the descriptor and xpubs need to be stored and saved by the user initially setting our any wallet so it is a "good known" descriptor to [...]

Yes, but from the point of view of privacy, generating a recipient address and exporting descriptors without using any software other than SS is a good idea.

They are just my opinions as a young user of SS.

Regards. -

Jaffasoft, 2025-03-19:
Apologies if i made a mess of this post it's bit late and not sure i formatted right new in this area! 

> 1. An option to disable the screensaver could be added to the advanced settings.
Cool. 

Select Screen Saver (only one though):
Advanced >> 
Static ✅
Animated ✅
None ✅
Own Massage ✅

Would be good or maybe there is other ideas too. 

> 3. Marking a fingerprint as used, I think, would be useful to do with a txid, as you might want to sign several PSBTs.> 
> [image] (sorry for my poor art)
Learn something new everyday. I don't know fully in depth under the hood how it might be implemented. I just offering the suggestion from a user point of view where i confused if i had use one to sign or not. If txid is the way then great. I guess could be a clear button or such? Or stacks a few different PSBT i'm not sure?? Perhaps opens pandora's box, 

> I think a way to unmark would also be desirable. Additionally, this mark could be necessary for single-sign PSBTs—why not?
Mmm, yeah i guess it gets tricky in different scenarios. Yes, least single sig as a start would be good. Since this is most common and adequate security for most.
  
> 3. 'Create Wallet Address': Do you mean generating a multisig address without a wallet descriptor? If so, you would need to select the script and derivation path as additional steps.

Yes for instance you don't have any wallet, never created one, don't want to go online at all. Have a SeedSigner, rolled die offline input your 24 or 12 words create the checksum (all these functions are in SS) and press the Lookup Address or Create Wallet Address' so you can then pay into that address without going online at all. Only thing you'd need to do is get the address QR code from SS and put it into the website on exchange or maybe scan QR into your phone copy paste the address into the exchange pay the BTC into that address. Transaction done 100% offline. Ppl stacking BTC not intend spending scenario. The same address can be checked loading into a address explorer online to see the transaction occurred. 

Yes good point seed signer would need to generate a descriptor, using derivation path and script etc some input i guess that complicates it a bit. Maybe the reason its not a feature now. Though also maybe it's from a security point of view i.e getting ppl to print the descriptor and or xpubs to save and store all this info on paper or metal. 
> This must be the user's responsibility.
For a brand new wallet its most probably to pay in the very first address. For subsequent addresses 

> > Doing this it might as well have the function to put a single sig or Multisig wallet descriptor up on screen QR code so it can go into a wallet sometime as well.

> Do you mean something like an xPub export option but for multisig wallets? In other words, exporting the wallet descriptor created in the feature mentioned in point 3?
Yes also the option to simply export the created descriptor in QR. User might do that anytime into the future making more time offline. To do a Multisig now take a bit of effort, each seed is loaded then each xPub is exported out to say Sparrow Wallet. Even just for that function alone (making a descriptor in SS) is going to be convenient. Maybe all this can't be done due to QR code amount of bits they can carry. I asked GROK3 and it said single sig descriptor could fit on a 29 x 29 QR code. 

> Yes, but from the point of view of privacy, generating a recipient address and exporting descriptors without using any software other than SS is a good idea.
>
> They are just my opinions as a young [...]

> Regards. -

Same, but i probably don't realise fully the coding and hard work goes into just a simple idea and is it really worth it. And since its almost inevitable to create an online wallet sometime to view all the transaction, UTXO management, plus store backup when first created etc it might not be worth the trouble. 

I would really like to be able to turn that animation off though and know which fingerprints have signed already.

kdmukai, 2025-03-19:
@Jaffasoft to give your input the best chances of being addressed / implemented, I strongly recommend that you split off each separate item into its own Github Issue. You can refer back here in those new Issues by just saying something like:

"This was originally discussed in Issue #692" (notice that Github auto-links "#" reference numbers).

You can then close this Issue. Closing does not delete an Issue, so the "#" references will still be valid.

Keeping things separate makes it easier for a developer to note things like "This proposed change addresses Issue #xyz...". But obviously if Issue #xyz is talking about a number of different things it just creates a bit of a mess. Also makes it harder to ensure that none of your items get lost (e.g. "I fixed #xyz so that's now closed and we never have to look at it again" except only one aspect of #xyz was actually addressed).

closed 2025-03-19 by Jaffasoft