SeedSigner/seedsigner #693: [Enhancement] Customize `gettext` with more robust error handling

kdmukai, 2025-03-23:
Some complex translation strings include named variables that must be included in the translation strings for each language.

e.g.
```python
# TRANSLATOR_NOTE: Final word calc. `mnemonic_length` = 12 or 24. `num_bits` = 7 or 3 (bits of entropy in final word).
text=_("The {mnemonic_length}th word is built from {num_bits} more entropy bits plus auto-calculated checksum.").format(mnemonic_length=self.mnemonic_length, num_bits=self.num_entropy_bits)
```

But the current German translation is:
```python
{}es Wort wird erstellt aus {} mehr Entropiebits plus errechneter Prüfsumme
```

Where the empty brackets `{}` aren't compatible with the newer named variable approach.

For clarity, the translation string should be:
```python
{mnemonic_length}es Wort wird erstellt aus {num_bits} mehr Entropiebits plus errechneter Prüfsumme
```

`gettext` raises the following error in this case:
```
IndexError: Replacement index 0 out of range for positional args tuple
```

That is obviously not the most helpful nor intuitive explanation.

## Initial approach (meh)
I started building a custom Exception class with some manual fields to set:
```python
class TranslationVariableMissingError(Exception):
    """ Raised when a translation does not include the expected named variable(s).

        e.g. source string: "Hello, {name}!" but the translation is "Bonjour!" or
        "Bonjour, {}!" instead of "Bonjour, {name}!"
    """
    def __init__(self, *args, expected_named_variables: list[str] = None, **kwargs):
        # Importing here to avoid circular import issues.
        from seedsigner.models.settings import Settings
        from seedsigner.models.settings_definition import SettingsConstants
        super().__init__(*args, **kwargs)
        self.expected_named_variables = expected_named_variables
        self.locale = Settings.get_instance().get_value(SettingsConstants.SETTING__LOCALE)

    def __str__(self):
        if self.expected_named_variables:
            return f"{self.__class__.__name__}: '{self.locale}' translation likely missing {', '.join([f'`{var}`' for var in self.expected_named_variables])} named variables in the translation string."
        return f"{self.__class__.__name__}: locale: {self.locale} | {super().__str__()})"
```

And then used in the main code like:
```python
        try:
            # TRANSLATOR_NOTE: Final word calc. `mnemonic_length` = 12 or 24. `num_bits` = 7 or 3 (bits of entropy in final word).
            text=_("The {mnemonic_length}th word is built from {num_bits} more entropy bits plus auto-calculated checksum.").format(mnemonic_length=self.mnemonic_length, num_bits=self.num_entropy_bits)
        except IndexError as e:
            traceback.print_exc()
            raise TranslationVariableMissingError(expected_named_variables=["mnemonic_length", "num_bits"]) from e
```

The problem with this approach is that we have to remember to wrap the `gettext` calls with the `try`/`except` custom handling every time. Plus repeating the expected named variables is not DRY, annoying, and a possible minor maintenance problem.

## Better approach
Extend `gettext` itself to always catch these exceptions and, if possible, automatically extract the expected named variables from the source string and populate the error message accordingly. This way, rather than having to remember the `try`/`catch` on each use and all the manual/custom error handling, it would all just automatically happen in our custom `gettext`.

In `seedsigner.helpers.l10n` we would extend `gettext` and then update all our imports:
```python
from seedsigner.helpers.l10n import gettext as _
```

### Additional error handling
Further testing should be done to also catch any cases where the wrong number of blank variable insertions are found in the translation string:
```
source string = "I like {}, {}, and {}."
translation string = "Me gusta {} and {}"
```

AdityaPandeyCN, 2025-04-01:
Hello @kdmukai Hope you are doing well I went through the issues and would like to work on it. I'm thinking of  implementing the enhanced gettext function as suggested in the issue. My plan is to:

1. Create a custom gettext wrapper in src/seedsigner/helpers/l10n.py that can handle variable formatting directly
2. Add clear error messages when translations are missing named variables (like when German has {} instead of {mnemonic_length})
3. Make it fall back to the English text with proper formatting when there's an error
4. Use monkey-patching so we don't have to change imports everywhere

What do you think of it should I proceed with PR.

kdmukai, 2025-04-01:
> 3\. Make it fall back to the English text with proper formatting when there's an error

I have two thoughts here. On the one hand, yes, having a clean fallback would ensure the best possible handling if this were to happen in a production release. Makes sense. Hard to argue against.

But on the other hand, our automated testing processes should find those sorts of errors way before they have any chance of getting to a release. And in the testing phase, we want this to be a showstopper exception. "ERROR! BLAH BLAH BLAH," the test fails or the screenshot generator exits with the error.

We could have this operate in two different contexts: local dev/testing vs production.

If production, fall back to the properly formatted English.

If local dev or testing, fail out with a big error.

But that all being said, this also feels a little bit like overkill. I say just let it fail spectacularly and we better catch it in time.

Also: even if we were to have the production mode fall back, we'd still have to be sure we write test scenarios to verify that the fall back works as expected. That could maybe get messy trying to juggle: "When we're testing, don't use the fall back, but during THIS test, pretend that we're NOT testing so we can test the production fall back..."

kdmukai, 2025-04-01:
Separate thought: we could think about having a test that explicitly goes through each translation string:
* If `n` replacement var `{}` appear in the source string, does the translated string have the same number?
* If a named variable appears in the source string, does the translated string include it as well?
  * And vice-versa: if a named variable is in the translated string but not in the source string, that's also bad.

kdmukai, 2025-04-01:
> 4\. Use monkey-patching so we don't have to change imports everywhere

I'm open to debate on this, but my instinct is that I would want it to be clear that we are using our own custom version of `gettext` (e.g. `custom_gettext`, `seedsigner_gettext`, ... I dunno). So I wouldn't want our version invisibly monkeypatched over the normal `gettext`.

In my proposed custom implementation, we'd be adding optional kwargs to `gettext`. If a new developer is unfamiliar with the history here and they search for how to use `gettext`, they'll be confused that they can't find any info about how the kwargs are used (because the original function doesn't use kwargs, only our custom version would).

Yes, such a change would touch many many files, but the code review wouldn't really be any harder since it's just a trivial import change.

AdityaPandeyCN, 2025-04-01:
Thankyou for the feedback, I see what you mean about making errors fail instead of hiding them with fallbacks. That makes a lot more sense for catching bugs early. I'll create a function called seedsigner_gettext instead of doing the monkey-patching thing. I think you're right that it's better to be clear about using our own custom function rather than hiding it.

kdmukai, 2025-04-01:
> Separate thought: we could think about having a test that explicitly goes through each translation string:
>
> * If `n` replacement var `{}` appear in the source string, does the translated string have [...]

Logged this as its own Issue here: https://github.com/SeedSigner/seedsigner/issues/700

kdmukai, 2025-04-01:
> I'll create a function called seedsigner_gettext

I don't know that I love that name. But I don't necessarily hate it, either! Usually naming things is really tough for me and I agonize over it maybe a bit more than I should.

kdmukai, 2025-04-02:
Very basic sanity check: Does the test succeed when running against the unmodified `gettext`?

The translations are loaded via a git submodule that might not be initialized in a local dev setup. I believe it's:

```
git submodule init
git submodule update
```

AdityaPandeyCN, 2025-04-02:
This worked, Thankyou for this also I have created a PR #702 Please Take a Look when time permits.