SeedSigner/seedsigner #700: [Testing / l10n] Test translation strings for proper replacement vars / named variable insertion

kdmukai, 2025-04-01:
Originally discussed in: https://github.com/SeedSigner/seedsigner/issues/693#issuecomment-2769822618

## Problem:
Transifex translators have to correctly insert `{}` variable placeholders in their translation strings as well as named variable `{some_var}` placeholders.

If they fail to include them, include the wrong number of `{}`, omit named variables, or include unexpected named variables, `gettext` will throw an error.

## Solution(?):
Add a test that explicitly goes through each translation string:
* If `n` replacement var `{}` appear in the source string, does the translated string have the same number?
* If a named variable appears in the source string, does the translated string include it as well?
  * And vice-versa: if a named variable is in the translated string but not in the source string, that's also bad.

The test can use simple string / regex tests comparing the English source string and the translated string.

Each translation *.po file should be pretty easy to parse:
```
#. current roll number vs total rolls (e.g. roll 7 of 50)
#: src/seedsigner/gui/screens/tools_screens.py
msgid "Dice Roll {}/{}"
msgstr "Würfeln {}/{}"

#. Build the last word in a 12 or 24 word BIP-39 mnemonic seed phrase.
#: src/seedsigner/gui/screens/tools_screens.py
msgid "Build Final Word"
msgstr "Finales Wort ermitteln"

#. Final word calc. `mnemonic_length` = 12 or 24. `num_bits` = 7 or 3 (bits of
#. entropy in final word).
#: src/seedsigner/gui/screens/tools_screens.py
msgid ""
"The {mnemonic_length}th word is built from {num_bits} more entropy bits plus"
" auto-calculated checksum."
msgstr ""
"{}es Wort wird erstellt aus {} mehr Entropiebits plus errechneter Prüfsumme"
```

abdallhfattah, 2025-04-24:
Thanks for explaining this issue. if I understand correctly, the problem is about variable placeholders in translations:

When the app gets translated, text with placeholders like `{}` or named variables like `{mnemonic_length}` needs to be handled carefully by translators. If they make mistakes with these placeholders (missing some, adding extra ones, or changing named variables to unnamed ones), the app will crash when it tries to run in that language.

Your proposed solution makes sense - creating an automated test that checks each translation to ensure:

- The same number of `{}` placeholders exist in both original and translated strings
- All named variables from the original text appear correctly in the translation
- No unexpected variables appear in the translation

This would help catch these errors before they cause runtime problems for users. I think this is a good approach to prevent these translation crashes.

Is there anything else about this issue I should understand?

kdmukai, 2025-04-24:
PR 702 would catch errors as they happen during execution, but would not pre-emptively detect errors which is more the goal of this Issue.

If 702 moves forward (I need to give the PR some overdue attention), I think it would make sense to leverage it in a new test that explicitly runs through every translation string in each language.

So I'd consider this issue a follow-up TODO item.

Also it's worth looking into the `compile_catalog` command itself to see if there's any kind of validation checking option we can enable. I'd be surprised if something wasn't already built in, but I haven't researched it.