SeedSigner/seedsigner #701: Remove belenaEtcher recommendation in favor of Rufus

tadeubas, 2025-04-01:
Recently in February 2025 Tails OS changed its recommendation to [use Rufus instead of balenaEtcher](https://tails.net/news/rufus/index.en.html). I believe SeedSigner could do the same [here for example](https://seedsigner.com/software/)

Dimi8146, 2025-04-01:
I favor Rufus. 

Honorable mention:
https://github.com/raspberrypi/rpi-imager

SeedSigner, 2025-04-02:
The privacy concerns with balena are a legit reason to look at other options. I lean towards Raspberry Pi Imager. The Rufus website feels a little scammy with meh design and having multiple ads loading; it also wasn't immediately clear to me if a version for MacOS was available. The RPi Imager download page is much cleaner, no ads, and versions for Windows, MacOS and Linux are right there. Open to other perspectives on this.

etkaar, 2025-10-24:
Since SeedSigner is such a sensitive piece of software I would—when it comes to Linux—recommend nothing more than a simple

```
sudo -s
cat seedsigner_os.0.8.6.pi0.img > /dev/sd[x] && sync
```

after validation is done. This is the most secure method. I see no reason for recommending any imaging software.