SeedSigner/seedsigner #715: [Feature request] Test: Linter

polespinasa, 2025-04-07:
Adding a Linter to the tests is a good idea to keep a good code format easy to read for review.
Perhaps the [Bitcoin Core](https://github.com/bitcoin/bitcoin/tree/master/test/lint) one can work as a reference.

Chaitanya-Keyal, 2025-04-07:
Hey, I've raised PR #707 which aims to resolve this.

It's currently being discussed and hopefully we can get a consensus for using `Black`.

kdmukai, 2025-04-09:
My current opinion is:
* I still prefer our slightly non-standard / opinionated approach to formatting.
* But I'm happy to respect the consensus view of our core contributors if most would rather shift to be more PEP-8 compliant.
* BUT MOSTLY: I just don't have the interest / energy / time to make this aesthetic debate a priority. It's just not worth our time right now (but because I dislike the change, I don't want to just say, "Okay, sure, go ahead").
* But I do see the value of an auto-formatter at least for consistency.

THEREFORE: I'd prefer the `yapf` option that can be customized to match our current non-standard style. Even better if it can disable the formatting on conditionals to preserve the cases where we've intentionally left conditions on one long line.

And table / defer the style discussion to sometime later.

So it would be a compromise: Ensure consistency based on where we are now. But lays the groundwork for easily shifting us closer to PEP-8 should we opt to do so later.

Chaitanya-Keyal, 2025-04-09:
Totally understand that — just wanted to add that I couldn’t find a way to prevent `yapf` from reformatting conditionals. So, if we do go ahead with setting it up, it _will_ reformat long conditionals, and honestly, I don’t think that improves readability in those cases.

To reiterate, the only real differences from our current (_only slightly_ non-standard) style that `black` would introduce are:
1. Long conditionals get split across lines (`black` does this in a much cleaner way than `yapf` IMO)
2. Line spacing:
    - Top-level definitions: 3 lines → 2
    - Class methods: 2 lines → 1
3. Minor changes for consistent formatting across the board

`yapf` also introduces (1) and (3), and while it lets us customize spacing to keep 3 and 1 lines (so it's a bit closer to our current style), it’s still not fully aligned. Plus, because it’s less strict than `black`, it wouldn’t fully eliminate formatting discussions in PRs either.

That said, I completely get deferring the decision for now. Maybe we could set up a quick reaction-based poll here or on Telegram to gauge all core contributor's preferences and then get back to it later on?

kdmukai, 2025-04-22:
More in-depth discussion in telegram:

https://t.me/seedsigner_new_devs/11503

https://t.me/seedsigner_new_devs/11533

https://t.me/seedsigner_new_devs/11545