SeedSigner/seedsigner #726: Recent model WaveShare hats have a different ST7789 screen

jdlcdl, 2025-04-11:
Today in the SeedSigner Community telegram group, User "BitK" shared info learned from speaking to WaveShare reps in Korea about late-model screens being too bright, then BitK shared a cut/paste of ST7789 code recomendations after referring WafeShare reps to review seedsigner's code base.

After white-space cleanup, the diff with changes and current release for src/seedsigner/hardware/ST7789.py is below

`diff -a --strip-trailing-cr /tmp/BitK-ST7789.20250411 ~/seedsigner/src/seedsigner/hardware/ST7789.py`

```diff
48c48
<         self.data(0x00)                 #self.data(0x00)
---
>         self.data(0x70)                 #self.data(0x00)
61c61
<         self.data(0x00) 
---
>         self.data(0x35) 
64c64
<         self.data(0x3F)
---
>         self.data(0x19)
73c73
<         self.data(0x0D)   
---
>         self.data(0x12)   
86,103c86,87
<         self.data(0xF0)
<         self.data(0x00)
<         self.data(0x02)
<         self.data(0x01)
<         self.data(0x00)
<         self.data(0x00)
<         self.data(0x27)
<         self.data(0x43)
<         self.data(0x3F)
<         self.data(0x33)
<         self.data(0x0E)
<         self.data(0x26)
<         self.data(0x2E)
< 
< 
<         self.command(0xE1)
<         self.data(0xF0)
<         self.data(0x07)
---
>         self.data(0xD0)
>         self.data(0x04)
104a89,95
>         self.data(0x11)
>         self.data(0x13)
>         self.data(0x2B)
>         self.data(0x3F)
>         self.data(0x54)
>         self.data(0x4C)
>         self.data(0x18)
107,110c98,107
<         self.data(0x16)
<         self.data(0x26)
<         self.data(0x43)
<         self.data(0x3E)
---
>         self.data(0x1F)
>         self.data(0x23)
> 
>         self.command(0xE1)
>         self.data(0xD0)
>         self.data(0x04)
>         self.data(0x0C)
>         self.data(0x11)
>         self.data(0x13)
>         self.data(0x2C)
112,115c109,115
<         self.data(0x19)
<         self.data(0x19)
<         self.data(0x31)
<         self.data(0x3A)
---
>         self.data(0x44)
>         self.data(0x51)
>         self.data(0x2F)
>         self.data(0x1F)
>         self.data(0x1F)
>         self.data(0x20)
>         self.data(0x23)
162c162
<         self._spi.writebytes2(pix) 
---
>         self._spi.writebytes2(pix)	
169c169
<         self._spi.writebytes2(_buffer) 
---
>         self._spi.writebytes2(_buffer)
```

kdmukai, 2025-04-19:
On my screen:

Original gamma curve (our current `ST7789` driver) is too dark. Gray is too dark, orange renders too reddish.
```python
        # Original settings
        self.command(0xE0)
        self.data(0xD0)
        self.data(0x04)
        self.data(0x0D)
        self.data(0x11)
        self.data(0x13)
        self.data(0x2B)
        self.data(0x3F)
        self.data(0x54)
        self.data(0x4C)
        self.data(0x18)
        self.data(0x0D)
        self.data(0x0B)
        self.data(0x1F)
        self.data(0x23)

        self.command(0xE1)
        self.data(0xD0)
        self.data(0x04)
        self.data(0x0C)
        self.data(0x11)
        self.data(0x13)
        self.data(0x2C)
        self.data(0x3F)
        self.data(0x44)
        self.data(0x51)
        self.data(0x2F)
        self.data(0x1F)
        self.data(0x1F)
        self.data(0x20)
        self.data(0x23)
```

The newer `st7789_mpy` driver ends up a bit too light, the orange gets a bit too washed out, but is a closer to match to our UX design target:
```python
    # uses a different command setting format
                                            # Set gamma curve positive polarity
    ( b'\xe0', b'\xd0\x00\x02\x07\x0a\x28\x32\x44\x42\x06\x0e\x12\x14\x17', 0),
                                            # Set gamma curve negative polarity
    ( b'\xe1', b'\xd0\x00\x02\x07\x0a\x28\x31\x54\x47\x0e\x1c\x17\x1b\x1e', 0),

# Converted to the ST7789 driver format:
        self.command(0xE0)
        self.data(0xd0)
        self.data(0x00)
        self.data(0x02)
        self.data(0x07)
        self.data(0x0a)
        self.data(0x28)
        self.data(0x32)
        self.data(0x44)
        self.data(0x42)
        self.data(0x06)
        self.data(0x0E)
        self.data(0x12)
        self.data(0x14)
        self.data(0x17)

        self.command(0xE1)
        self.data(0xd0)
        self.data(0x00)
        self.data(0x02)
        self.data(0x07)
        self.data(0x0a)
        self.data(0x28)
        self.data(0x31)
        self.data(0x54)
        self.data(0x47)
        self.data(0x0e)
        self.data(0x1c)
        self.data(0x17)
        self.data(0x1b)
        self.data(0x1e)
```

The curve in the "BitK" attachment above is even brighter with the orange rendering closer to yellow:
```python
        self.command(0xE0)
        self.data(0xF0)
        self.data(0x00)
        self.data(0x02)
        self.data(0x01)
        self.data(0x00)
        self.data(0x00)
        self.data(0x27)
        self.data(0x43)
        self.data(0x3F)
        self.data(0x33)
        self.data(0x0E)
        self.data(0x0E)
        self.data(0x26)
        self.data(0x2E)

        self.command(0xE1)
        self.data(0xF0)
        self.data(0x07)
        self.data(0x0D)
        self.data(0x0D)
        self.data(0x0B)
        self.data(0x16)
        self.data(0x26)
        self.data(0x43)
        self.data(0x3E)
        self.data(0x3F)
        self.data(0x19)
        self.data(0x19)
        self.data(0x31)
        self.data(0x3A)
```

---

A "goldilocks" gamma curve isn't quite present, but the `st7789_mpy` gets the closest for my screen.

We should probably offer at least these three curves to be options in Settings.

Nice to have: get a more precise middle curve for what we assume to be the majority of screens.

etkaar, 2025-10-24:
I encountered this issue today when I got two new Waveshare 1.3 inch HAT displays (ST7789). Expected would be this:

[image]

But now I get this, it looks really awful:

[image]

kdmukai, 2025-10-24:
The "_mpy" curve (green) renders lighter and more washed out vs the "Original" curve (blue) on an actual screen.

So these values may be inverted: lower value means less intensity reduction?

And the above is just the "positive" curve. According to Copilot: "The ST7789 uses separate gamma curves for positive and negative voltage polarities. LCD displays alternate the polarity of the voltage applied to pixels to prevent degradation, so both polarities need proper gamma correction to maintain consistent appearance."

etkaar, 2025-10-24:
I wonder how the values can be different for the same ST7789 chip (or whatever that stands for)?

kdmukai, 2025-10-24:
@jdlcdl looks like I'm about to start building a 14-param curve tweaker UI built into SeedSigner... 👀

A bit worried about the positive polarity curves being different from the negative polarity curve. If Copilot is right, I suppose if, say, the positive is tweaked but the negative isn't, you might expect to see flashing as the polarity flips and the curve applies different boost/dampen rules. If that's true, then each tweak will have to be like:

* Change a value
* Now change its negative polarity equivalent until the flashing goes away?

Just thinking out loud. This is 100% a FAFO situation.

alvroble, 2025-10-24:
For now what we only know is that there are some screens that are recommended to work with a set of values (our default), and some that are recommended to work with a different set (BitK ones, which came from Waveshare reps), and then the ones for mpy. I think that at some point we don't even know what the values stand for and what scale do they use. So maybe the best option is to go for a set of "recommended" presets?

It would be great if @waveshare could help us here.

kriswilk, 2025-10-25:
Thanks @kdmukai for visually comparing the three gamma curves. However I think there may be more going on here... looking at @etkaar's photos (https://github.com/SeedSigner/seedsigner/issues/726#issuecomment-3443252507), the difference is extremely dramatic. The black level is really a mid-gray.

This is exactly what I observed with the module I got. It's really quite terrible and not like any of the three samples @kdmukai pictured.

Beginning to wonder if these panels are actually defective? Hard to know without a larger sample.

kdmukai, 2025-10-25:
@kriswilk yes, so far I would not expect any of these curves to correct for what was shown above.

For example, the NTSC standard has something like 16 as its pure black value. But computer monitors treat 0 as pure black. So if you display unmodified NTSC-targeted content on a computer display, the black levels look washed out because the NTSC signal never reaches 0. That content needs a gamma curve that basically says "the displayable range is 16-245" and then the display device can map those values to its actual range (0-255).

Without knowing the exact mapping problem, I don't have much confidence in being able to guess and check a possible solution.

---

My work thus far is just the proof of concept to see if we can apply different curves at all (success!) plus a secondary finding of whether or not Copilot knows what it's talking about (apparently also a tentative yes!).

I probably can't troubleshoot these extreme cases until I get my hands on one myself.

And even if we can't yet correct the extremes, the more subtle variations in screens has been an annoyance for a while. And made worse by the SeedSigner+'s bigger 320x240 screen that uses a different ST7789 python driver which has a different curve, causing the SeedSigner+ to render differently from our baseline (a good Waveshare screen sample renders like the middle curve in the side-by-side comparison while the SeedSigner+ is obviously more like the "lighter" curve).

So if nothing else, this R&D will help us wrangle the good or good enough screens into better consistency.

etkaar, 2025-10-25:
I think if there is a "Waveshare 1.3 inch IPS Display HAT 240x240" with a ST7789 driver they absolutely must not behave differently, at least not if the difference is day and night. That doesn't make sense for me.

Therefore, the question is, if these displays shouldn't be treated defect instead. Help from @waveshare would be great.