SeedSigner/seedsigner #768: [Feature] Add BIP-353 proof validation

kdmukai, 2025-06-10:
# BIP-353 Proof Validation

see: [BIP-353](https://github.com/bitcoin/bips/blob/master/bip-0353.mediawiki)

see: [YouTube: Corallo discusses BIP-353 on What Bitcoin Did @ 1:03:45](https://youtu.be/0iMhYvu5KqU?t=3825)

Via Matt Corallo:

If the HWW identifies a DNSSEC proof in the PSBT (as defined at https://github.com/bitcoin/bips/pull/1657) it should verify it (eg using [https://docs.rs/dnssec-prover/](https://t.co/yXSK5bBCiY)) so that the HWW can display ₿user@domain (per https://github.com/bitcoin/bips/blob/master/bip-0353.mediawiki). Should be quite straightforward to do (on any HWW with enough free ROM to include the validator crate).

ie BIP 353 enables wallets to bring the LN-Address scheme (which users seem to love and has clearly found PMF) to on-chain payments (incl silent payments) and, much more importantly, it enables HWWs to provide something on-screen that a human can actually verify, unlike addresses. Given the common use-case of a user sending funds from their HWW to an exchange, where they verify the address displayed on their (malware-compromised) computer matches the one on their HWW ensures HWWs provide zero security, this is a massive step-up since users can actually verify that they're sending the money to ₿myuser@river.com.

kdmukai, 2025-07-30:
Update:
https://x.com/TheBlueMatt/status/1948718509275709666

Two software wallets now support bip-353: Cake and Sparrow.

alvroble, 2025-08-06:
Hi @Bicaru20. Please check https://github.com/diybitcoinhardware/embit/pull/102 as there is already an implementation proposal using the Rust library.

alvroble, 2025-08-06:
@Bicaru20 It might be valuable to explore an alternative proposal to https://github.com/diybitcoinhardware/embit/pull/102, using the Python ports available [here](https://github.com/alvroble/pydnssec-prover/tree/main). This could motivate broader discussion around which approach best serves not only `embit` and `SeedSigner`, but also downstream HWW projects relying on `embit`.

One thing that made me try with the Rust implementation is that `embit` currently doesn't have any external Python library requirements. Only depends on `secp256k1`  which is covered by C bindings and a fallback Python implementation.

Bicaru20, 2025-08-07:
> [@Bicaru20](https://github.com/Bicaru20) It might be valuable to explore an alternative proposal to [diybitcoinhardware/embit#102](https://github.com/diybitcoinhardware/embit/pull/102), using the Python ports available [here](https://github.com/alvroble/pydnssec-prover/tree/main). This could motivate broader discussion around which approach best serves not only `embit` and `SeedSigner`, but also downstream HWW projects relying on `embit`.
> 
> One thing that made me try with the Rust implementation is that `embit` currently doesn't have any external Python library requirements. Only depends on `secp256k1` which is covered by C bindings and a fallback Python implementation.

I tried both implementation. I liked more the Python implementation, because when you call the verification function, it returns a dictionary and in one of the field of that dictionary you could see the dns_name:
`'{"valid_from": 1754011162, "expires": 1754394592, "max_cache_ttl": 3600, "verified_rrs": [{"type": "txt", "name": "craig.user._bitcoin-payment.sparrowwallet.com.", "contents": "bitcoin:bc1qwthe43xeuasklclq4kvhreluv3hu92rzej42js"}]}'`

Mean while when I use the embit library which uses the rust implementation I get: 
`{'valid_from': 1754011162, 'expires': 1754394592, 'max_cache_ttl': 3600, 'verified_rrs': []}`

Both returns the proof as valid, however I am worried (probably for my lack of knowledge in this field) that a user can change the `dns_name` and use a valid proof and would still be valid because the validation function does not compare if the `dns_name` corresponds to the proof passed. 

For example, this two verification with the same proof are both valid:

`verify_dns_proof(''craig@sparrowwallet.com.', proof)`

`verify_dns_proof(''seedsigner@sparrowwallet.com.', proof)`

With the python implementation, since in the dictionary returned there is the name, I could double-check to ensure the `dns_name` correspond with the proof. With the rust implementation I cannot or at least I don't know how to do it.

alvroble, 2025-08-07:
> Mean while when I use the embit library which uses the rust implementation I get:
> `{'valid_from': 1754011162, 'expires': 1754394592, 'max_cache_ttl': 3600, 'verified_rrs': []}`

@Bicaru20 Can you share more details of the platform/environment you used when testing this?

I do get the `verified_rrs` from the embit library, you can check the test vectors used [here](https://github.com/diybitcoinhardware/embit/blob/08bf174136e04710dedd0b7767b41806b65162b4/tests/tests/test_bip353.py#L16-L19). But I'm curious on under which conditions you got that result.

Currently my python implementation does not pass the new invalid proofs test vectors that were proposed in https://github.com/bitcoin/bips/pull/1912. So I'd say is still soon for it to be production-ready.

alvroble, 2025-08-07:
> But I don't know why in the `pydnssec_prover` works both options.

Probably another bug 😊. As I said the Python library is a port and should have more testing and reviews, if it does serves a the use case.

Bicaru20, 2025-08-07:
> Looking at the proof, you might conclude the user is paying to `matt@mattcorallo.com` (and the proof is valid for such a name lookup!) but actually they wanted `a.x_domain_cname_wild@dnssec_proof_tests.bitcoin.ninja`. Thus you need to validate against a specific name and see if the proof allows for that name, rather than being able to look up the name from the proof.
@TheBlueMatt what I'm currently doing is to look in the `verified_rrs` field inside the dictionary the validation returns. There I look for some element to have the same `name` as the one in the psbt. I do not look if it is `TXT` or `CNAME` as all I want is for some element with the same `name`. Is this approach correct?