SeedSigner/seedsigner #775: Improve Back Navigation Logic in ScanSeedQRView

Advaitgaur004, 2025-06-19:
There seems to be a structural issue with how back navigation is handled in the [ScanSeedQRView](https://github.com/SeedSigner/seedsigner/blob/2fe92ce1ef613fe777d44767264c4ab2ef69d141/src/seedsigner/views/scan_views.py#L183), which inherits from [ScanView](https://github.com/SeedSigner/seedsigner/blob/2fe92ce1ef613fe777d44767264c4ab2ef69d141/src/seedsigner/views/scan_views.py#L14).

In the run() method of ScanView, the RET_CODE__BACK_BUTTON case isn’t explicitly handled. As a result, if none of the branches match, it defaults to returning to [MainMenuView](https://github.com/SeedSigner/seedsigner/blob/2fe92ce1ef613fe777d44767264c4ab2ef69d141/src/seedsigner/views/scan_views.py#L169), regardless of the user’s previous context.

Example of Confusing UX Flow:

- Tools  -> Address Explorer -> Scan a Seed -> Back ->  lands on Main Menu [It should land on Address Explore Menu]
- Tools -> Verify Address -> Back ->  Main Menu, ahhh [It should land to Tool Menu]
- Seeds -> Scan a SeedQR -> Back ->  Main Menu again [It should land to Seeds Menu]

This creates and unexpected navigation flow, which can confuse users who expect to go back to the previous tool or menu.

### One quick solution that I thought of:

Let any view that inherits from ScanView specify a back_destination, so that pressing the back button takes the user to the actual previous or expected screen, rather than always defaulting to MainMenuView
```python
class ScanView(View):
    def __init__(self, back_destination=None):
        super().__init__()
        self.back_destination = back_destination
```

and later on inside the run() method of ScanView, update the fallback block like this:
```python
if ret == RET_CODE__BACK_BUTTON:
    if self.back_destination:
        return Destination(self.back_destination)
    else:
        from seedsigner.views.menu_views import MainMenuView
        return Destination(MainMenuView)
```

For example, 

```python
from seedsigner.views.scan_views import ScanSeedQRView
from seedsigner.views.tool_views import AddressExplorerMenuView

return Destination(ScanSeedQRView, view_args={"back_destination": AddressExplorerMenuView})
```

Advaitgaur004, 2025-06-21:
More similar flows with back navigation issues.

Seeds Menu -> Enter 12-word seed -> Back -> Main Menu [It should land to Seeds Menu]
Seeds Menu -> Enter 24-word seed -> Back -> Main Menu [It should land to Seeds Menu]

abhigit-saha, 2025-06-21:
> More similar flows with back navigation issues.
> 
> Seeds Menu -> Enter 12-word seed -> Back -> Main Menu [It should land to Seeds Menu] Seeds Menu -> Enter 24-word seed -> Back -> Main Menu [It should land to Seeds Menu]

I don't think this seems to be a problem. When I press back at the i'th seed phrase word, I do get routed back to i-1'th one, and that is expected behaviour. Other screens should route it back to the main menu. Do let me know if I am missing some screens where the behavior is weirder.