SeedSigner/seedsigner #777: Document pytest test running instruction, ability to run the tests on a dev machine

kornpow, 2025-06-20:
Trying to run the tests for the code, I noticed that it isn't specifically documented how to run the tests.

At the moment, it also seems like the only way to run the tests is if you are on a Raspberry Pi, because of the `PiCamera` dep. That could be documented.

Things that might be useful:

1. Create a `requirements-dev.txt` which contains `pytest` and and other tools we may want to use. `black` has been suggested at times, though controversial, `flake8` could be good as well.
2. Even on the RPi, and have manually installed `pytest` I still need to do this to run the tests `PYTHONPATH=src pytest`

I might try and tackle parts of this, but wanted to create an issue to remind me.

Chaitanya-Keyal, 2025-08-28:
If I’m not mistaken, the test suite is designed to run on non-Raspberry Pi hardware as well. The documentation can be found here: [/tests/README.md](https://github.com/SeedSigner/seedsigner/blob/dev/tests/README.md). There’s also a [/tests/requirements.txt](https://github.com/SeedSigner/seedsigner/blob/dev/tests/requirements.txt).

That said, I agree it would be better to consolidate this into a `requirements-dev.txt` and add other developer tools like linters/formatters. #715 is related to this. Personally, I’d love to see `black` integrated into the codebase.

kornpow, 2025-08-30:
I've been getting really into the `uv` python package management ecosystem recently. Its been a game changer since it can manage the python environment in addition to the python installation as a whole.

The same organization makes a tool called `ruff` which does linting and also formatting. 
https://github.com/astral-sh/ruff

With `uv` we wouldn't really need any requirements.txt or requirements-dev.txt, all the dependancies are defined in `pyproject.toml` and can be grouped.

Ruff supports a config file, so we could potentially format it our own custom way, since I know @kdmukai wasn't a huge fan of the black formatter

Chaitanya-Keyal, 2025-08-30:
I agree! I love `uv`. In fact, I _just_ mentioned it in #808.

`ruff` is also pretty nice, although it’s designed to be a drop-in replacement for `black`. You’re right that it supports a config file, but the options are extremely limited, just like `black`. The available configurations are listed in their [docs](https://docs.astral.sh/ruff/settings/#format) and imo they aren't anything we'd really want to change from the defaults.